microsoft/Windows-driver-samples #374: Source port 0 for UDP data while doing connect redirect FWPM_LAYER_ALE_CONNECT_REDIRECT_V4

puran157, 2019-05-15:
I get source port as 0 in WFPSampler for UDP redirection (0.0.0.0:0 -> 192.168.1.5.80) and actual redirection never happens, i.e I don't receive the udp connection on user mode application which is listening on redirected port.

puran157, 2019-05-16:
I am getting the source address/port at https://github.com/microsoft/Windows-driver-samples/blob/master/network/trans/WFPSampler/sys/ClassifyFunctions_ProxyCallouts.cpp using
`FWPS_CONNECT_REQUEST* pConnectRequest = (FWPS_CONNECT_REQUEST*)(pRedirectData->pWritableLayerData);
			SOCKADDR_STORAGE originalDestination;
			RtlCopyMemory(&originalDestination, &(pConnectRequest->remoteAddressAndPort), sizeof(SOCKADDR_STORAGE));
SOCKADDR_STORAGE originalSource;
			RtlCopyMemory(&originalSource, &(pConnectRequest->localAddressAndPort), sizeof(SOCKADDR_STORAGE));`

Its just the originalSource which has address as 0.0.0.0 and port 0. Also once the destination port is modified, I don't see UDP received on local UDP server. I did not observed this behaviour for TCP (i.e source port is valid there and redirection is working)

puran157, 2019-05-16:
I also tried getting it as `FWP_VALUE* localport = KrnlHlprFwpValueGetFromFwpsIncomingValues(pClassifyValues,
			&FWPM_CONDITION_IP_LOCAL_PORT);`, Still the port value is 0 and since the `ClassifyProxyByALERedirect` is first function called on filter match, it seems a bug on OS side. Please let me know your thoughts on this.

sagizar, 2021-05-25:
I'm also having the exact same issue. anyone have found a solution for this?

puran157, 2021-08-03:
Yup, you should add another condition in filter -> SourcePort greater than 0 and that should ensure you get proper traffic with Src port

closed 2021-08-03 by puran157