microsoft/Windows-driver-samples #377: Virtual audio driver for surround sound.

westside, 2019-05-23:
I'm not sure this issue should go to stackoverflow, not here.

I want to make a virtual audio driver which is supporting 7.1 surround like below. 
[image]

Is there any example for this? (even class name or example file.) 
I look through sysvad project, but could not find it.

levanmanhbka, 2019-09-25:
Hi!
I run sysvad example. But cannot install main driver.
Thank you very much.

lavilycn, 2022-09-02:
> I'm not sure this issue should go to stackoverflow, not here.
> 
> Anyway, I want to make a virtual audio driver which is supporting 7.1 surround like below. [image]
> 
> Is there any example for this? (even class name or example file.) I look through sysvad project, but could not find it.

Did you solve this problem later? Thank you

closed 2025-11-11 by v-junyli