microsoft/Windows-driver-samples #390: Network Driver Samples Test Signing Fails Always

NoelStephens, 2019-07-14:
I used to be able to turn off digital sig checking and set test signing on in order to try out various driver examples.  Been developing my own driver and up to date have had no problems...until I updated my driver samples repository with the latest updates....and all of the windows updates included.  (** note ** the driver I am developing generates a catalog file)
So my target kernel debug machine's boot loader settings are:
--------------------------------------
Windows Boot Loader
--------------------------------------
identifier                      {current}
device                          partition=C:
path                              \Windows\system32\winload.exe
description                   Windows Server
locale                            en-US
loadoptions                  DDISABLE_INTEGRITY_CHECKS
inherit                           {bootloadersettings}
recoverysequence        {10c0bd3a-51b1-11e9-836b-c80bfe18dbc2}
displaymessageoverride  Recovery
recoveryenabled           Yes
testsigning                    Yes
allowedinmemorysettings 0x15000075
osdevice                       partition=C:
systemroot                   \Windows
resumeobject               {10c0bd38-51b1-11e9-836b-c80bfe18dbc2}
nx                                OptOut
debug                         Yes
--------------------------------------
As you can see I have all of the appropriate/recommended settings (above) to install a test signed driver, but....
When I try to install any compiled NDIS driver sample (with a test certificate and properly signed) it always gives me the "The third party INF does not contain digital signature information" error now.  I have tried this compiling both x86 and x64 bit versions.  

The caveat is that if you build a project with a catalog file, then it seems to be happy...??

Some of the examples (i.e. the NDIS MUX example) complain about all sorts of issues when trying to generate a catalog file, and as such (unless you sign it with a valid signature through the windows dev center) it seems like this is currently broken?

I ran into this yesterday when I compiled and tested the ndis filter driver sample, but then just got it to generate a catalog file and it installed fine... 

Is there some known issue regarding test signed drivers not installing without a catalog file?
If so, is there a work around to get the MUX driver sample working with a test signed certificate?

Thanks!

rodwiddowson, 2019-07-14:
Can't help, but you might try asking in [NTDEV](https://community.osr.com/categories/ntdev).  They guys who hang out there speak driver signing.

NoelStephens, 2019-07-14:
> Can't help, but you might try asking in NTDEV.  They guys who hang out there speak driver signing.

Yeah, I have posted several questions there with little success of getting replies...or rather...when I get a reply it rarely is an actual solution and more often some links to the Windows Drive Development site which I have almost memorized at this point...and some of the API references were last updated between 2012-2013 when the Win8 and WS 2012 updates hit...and they still contain references to callout filter actions that have notes "only works under Windows 8 and Windows Server 2012...which is not true because they all work under current versions of windows and windows server today.

Anyway, if anyone has any tips or tricks on how to actually install test signed NDIS driver examples without generating a catalog file... please feel free to repond!  :)

NoelStephens, 2019-07-14:
As an update, got a catalog file building (included the _mp.inf) but now getting an error:
0xE00002F when following the instructions for installation and adding it as a protocol.
So, there is definitely something that has changed that makes installing a test signed NDIS MUX example driver broken at this point in time on Window 10 (1809) and Windows Server 2019 (1809) with all updates applied.

Just FYI.

NoelStephens, 2019-07-14:
I also confirmed this same behavior using Visual Studio 2019 and the 1903 WDK.  The previous attempts were with VS 2017 and the 1809 WDK.

NoelStephens, 2019-07-14:
Ok,
For the NDIS MUX example you must do the following to get past the "The third party INF does not contain digital signature information" error:

1.) Add both the muxp.inf and mux_mp.inf files to the project you are going to build (VLAN or NOVLAN)
2.) Edit both inf files by adding under the [Version] section: CatalogFile	= mux.cat
3.) Make sure you have already built the debug version of the notifyob project.
4.) Add the mux.dll file to the resources to be included in the catalog file
5.) Make sure you have a valid test signature set.
6.) Build the project in question (NOVLAN or VLAN)
7.) Copy both inf files to the output directory of the build (i.e. mux\driver\60\novlan\x64\Debug\mux)
8.) Copy all of the files in the output directory (i.e. mux\driver\60\novlan\x64\Debug\mux) to your target machine.
9.) Make sure your test certificate is signed.
10.) Follow the install instructions outlined in the MUX README.md file.

If you did everything correctly, then it should install properly...you should get two "warnings" about the signature but it should properly install at that time.

When I get a chance I will loop back and see if the projects have been updated for this new debug driver install behavior, otherwise I will try to clean up what I have working a bit to make it "automatic" as opposed to having to make the above modifications.

So...based on this behavior...I am going to say that you have to have everything inside a catalog file that is signed by your test certificate or it will give you the "invalid signature (0xE00002F) error".