microsoft/Windows-driver-samples #397: XPSDrvSmpl sample project fails at printer driver installation step

royyin224, 2019-08-07:
I downloaded the XPSDrvSmpl project and followed the steps to build the driver.

The driver is built without error but when I try to install it, it pops an error message:

[image]

I intuitively copy following dll files from {project root}/x64/Debug/package to {project root}/install folder,

1. xdbook.dll
2. xdcolman.dll
3. xdnup.dll
4. xdscale.dll
5. xdsmplui.dll
6. xdwmark.dll

I tried the installation again, this time the file not found error disappeared but the installation silently failed without any error message, returning to following screen

[image]

Did I miss any step in this process?

royyin224, 2019-08-07:
I was able to install the sample driver after I follow the instruction in this post to disable driver signature enforcement, I am closing the issue, sorry for the trouble

closed 2019-08-07 by royyin224

arsu-leo, 2020-02-25:
> I was able to install the sample driver after I follow the instruction in this post to disable driver signature enforcement, I am closing the issue, sorry for the trouble

Which post? :D

royyin224, 2020-02-26:
> > I was able to install the sample driver after I follow the instruction in this post to disable driver signature enforcement, I am closing the issue, sorry for the trouble
> 
> Which post? :D

Oops, sorry, I forgot to put the link in there, I think this was the [post](https://www.howtogeek.com/167723/how-to-disable-driver-signature-verification-on-64-bit-windows-8.1-so-that-you-can-install-unsigned-drivers/)

Prakash19921206, 2021-08-22:
@royyin224 ,
its been quite a while since this issue was opened.

having issue with latest wdk, sdk and windows even after disaabling driver signature enforcement.
details mentioned in [this ](https://github.com/microsoft/Windows-driver-samples/issues/539#issuecomment-903318865)issue

its not showing any error message/code when installation fails!
is there any work around for this?

Thanks & Regards
Prakash