microsoft/Windows-driver-samples #430: passThrough.inf不符合VS2019(16.3.4)规范

kouzhudong, 2019-10-11:
1>------ 已启动全部重新生成: 项目: passThrough, 配置: Debug x64 ------
1>Building 'passThrough' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
1>Stamping x64\Debug\passThrough.inf
1>Stamping [Version] section with DriverVer=10/11/2019,9.28.26.113
1>D:\code\git\microsoft\Windows-driver-samples\filesys\miniFilter\passThrough\passThrough.inf : error 1420: [DefaultInstall]-based INF cannot be processed as Primitive.
1>D:\code\git\microsoft\Windows-driver-samples\filesys\miniFilter\passThrough\passThrough.inf(25-25): error 1421: Section [DefaultInstall] should have an architecture decoration.
1>D:\code\git\microsoft\Windows-driver-samples\filesys\miniFilter\passThrough\passThrough.inf(36-36): error 1421: Section [DefaultUninstall] should have an architecture decoration.
1>D:\code\git\microsoft\Windows-driver-samples\filesys\miniFilter\passThrough\passThrough.inf(72-72): warning 1205: Section [MiniFilter.DriverFiles] referenced from DelFiles and CopyFiles directive.
1>已完成生成项目“passThrough.vcxproj”的操作 - 失败。
========== 全部重新生成: 成功 0 个，失败 1 个，跳过 0 个 ==========

HollisTech, 2019-11-16:
The minifilter samples are all broken for VS2019 builds because the inf file DefaultInstall section does not conform to new requirements.

lapaz17, 2022-03-01:
Any progress on this? Should I use VS 2022?