microsoft/Windows-driver-samples #456: scanner.c always sends to first chunk data to UserMode

pvthuyet, 2020-02-10:
```
            notification->BytesToScan = min( Data->Iopb->Parameters.Write.Length, SCANNER_READ_BUFFER_SIZE );

            //
            //  The buffer can be a raw user buffer. Protect access to it
            //

            try  {

                RtlCopyMemory( &notification->Contents,
                               buffer,
                               notification->BytesToScan );

            } except( EXCEPTION_EXECUTE_HANDLER ) {
```
If the data file is larger than SCANNER_READ_BUFFER_SIZE, the driver always sends data buffer from 0->SCANNER_READ_BUFFER_SIZE .
Does anyone have solution for this???

hsebs, 2020-02-11:
scanner.c read only the first sector of a file.
If you want to check whole file, add loop.

pvthuyet, 2020-02-12:
Thank you @hsebs,

closed 2025-11-11 by v-junyli