microsoft/Windows-driver-samples #462: Obcallback

jammgit, 2020-02-27:
ObRegisterCallbacks can not protect process from taskmgr killing.

[image]

 if (PreInfo->KernelHandle != 1)   // test FALSE
{

}

callback from kernel mode.

kirubaKaranT, 2021-05-17:
did you find any solution to prevent it ?

> 
> 
> ObRegisterCallbacks can not protect process from taskmgr killing.
> 
> [image]
> 
> if (PreInfo->KernelHandle != 1) // test FALSE
> {
> 
> }
> 
> callback from kernel mode.

kirubaKaranT, 2021-05-26:
> 
> 
> ObRegisterCallbacks can not protect process from taskmgr killing.
> 
> [image]
> 
> if (PreInfo->KernelHandle != 1) // test FALSE
> {
> 
> }
> 
> callback from kernel mode.

hi i found Solution for that issue 
reset/remove all critical Access ( like Windows Protected Process )
https://docs.microsoft.com/en-us/windows/win32/procthread/process-security-and-access-rights
https://docs.microsoft.com/en-us/windows/win32/procthread/thread-security-and-access-rights

closed 2025-11-11 by v-junyli