microsoft/Windows-driver-samples #468: Loopback mode doens't work on any rendering device

kkxiong, 2020-03-12:
I have built and installed the SYSVAD (with APO extension) devices on my WIndows 10 laptop.
When I tried to enabled the loopback capturing mode on any rendering devices (Speakers, Headhpones, SPDIF Interface) to mirror the rendered audio back to some other audio deivces, I always heard tone rather than any audio being played back.

How can I enable loopback capturing on the audio devices in the example driver project? I think loopback is a common feature on most of the rendering audio devices so it will be appreciated if this feature can be enabled in the example project.

rhymu8354, 2020-08-20:
I was just looking at the code trying to reverse-engineer this.  It turns out the loopback endpoints are programmed the same way as the other capture endpoints: they simply generate a constant sine-wave.  See `minwavertstream.cpp` lines 354-363:

```c++
            if (m_pMiniport->IsLoopbackPin(Pin_))
            {
                //
                // Loopbacks pins use a different frequency for test validation.
                //
                toneFrequency = m_ulLoopbackCaptureToneFrequency;
                toneAmplitude = m_dwLoopbackCaptureToneAmplitude;
                toneDCOffset  = m_dwLoopbackCaptureToneDCOffset;
                toneInitialPhase = m_dwLoopbackCaptureToneInitialPhase;
            }
```

PiliposyanDavid, 2025-01-09:
Hello @kkxiong  @rhymu8354 
Have you developed a loopback audio device? I’m also on it now. If yes, can I ask some help ? 

Thank you !