microsoft/Windows-driver-samples #488: How to build avscamera?

eanders-MS, 2020-04-27:
I'm trying to build avstream/avscamera, and getting errors. Can you please tell me what I'm missing in my build setup? Here's my setup:

* Windows 10 Enterprise
* Visual Studio 2019
* C:\Program Files (x86)\Windows Kits\10\Include\10.0.18362.0

When I build the project, I get lots of errors due to missing include files. This seems to be due to the value of NTDDI_VERSION being explicitly set in Common.h to NTDDI_WIN10_VB. When I comment that out, the build gets further, but dies on missing`ExAllocatePoolZero`. I change instance of that call to `ExAllocatePoolWithTag`, and get further. Next error is missing identifier: `KSPROPERTY_CAMERACONTROL_EXTENDED_RELATIVEPANELOPTIMIZATION`, and a few similar symbols.

It seems like I'm not on the right version of the driver kit, or something. Any ideas?

boger047, 2020-04-27:
seems you should install a pre release SDK such as 19041

eanders-MS, 2020-04-27:
Ok, I installed 19041 but still seeing the same undefined symbols. How can I check which SDK I'm using? Sorry new to this. Thank you.

eanders-MS, 2020-04-27:
It's not showing up in the project properties page, but is installed on disk.

Property page:
[image]

File explorer:
[image]

How can I get VS2019 to see the new SDK?

eanders-MS, 2020-04-27:
Possible progress: After installing 10.0.19608.0 from .iso, it showed up in Visual Studio Installer, but showed as not installed. So I let Visual Studio Installer install it. After that, it showed up in the project's Property Page. I set the project to use the preview SDK and rebuilt, but got this error:

> 1>C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(181,5): error : A WDK corresponding to target version '10.0.19041.0' was not found. Please install this WDK version before building.

Am I even on the right track here?

eanders-MS, 2020-04-27:
Oh maybe I went too far into the future. 10.0.19608.0 > 10.0.19041.0

boger047, 2020-04-27:
sorry. WDK need also