microsoft/Windows-driver-samples #498: 2004 Updates

EatonZ, 2020-05-12:
[This](https://github.com/microsoft/Windows-driver-samples/commit/96eb96dfb613e4c745db6bd1f53a92fe7e2290fc) is the last update for some drivers, such as [fastfat](https://github.com/microsoft/Windows-driver-samples/tree/master/filesys/fastfat).

Windows 10 2004 is about to come out. I think it's a good time to freshen up the examples based on the latest 2004 codebase.

EatonZ, 2022-07-23:
@NeoAdonis I noticed you [published an update](https://github.com/microsoft/Windows-driver-samples/releases/tag/win11-22h2), but there's nothing new? Just an updated readme file - was there a glitch uploading all the rest of the files?

closed 2025-11-11 by v-junyli

v-junyli, 2025-11-11:
@EatonZ sure I reopened the issue. But I think this version of WDK is too old that is no longer supported.

@prashantchahar can you comment?

EatonZ, 2026-04-08:
@prashantchahar