microsoft/Windows-driver-samples #531: Mouse cursor texture unexpectedly shown on Iddcx surfaces when hardware cursor is set up

alexpacio, 2020-07-23:
Hi,

if I decide to handle the cursor through hardware using the function IddCxMonitorSetupHardwareCursor() and I avoid to draw it in the surface, I'm currentlly facing a weird behaviour: the cursor is not drawn as expected in the righ way, except if I drag things on the surface (eg. when dragging windows you'll see appear the mouse pointer drawn in the cursor position and, when the mouse click is released, it disappears again).

Is it caused by something that I'm missing in the mouse cursor handling (honestly I don't think so, I debugged the problem enough to say that I'm pretty sure that I'm doing it in the righ way) or is it a bug with Iddcx?

pccq2002, 2020-10-09:
IddCxMonitorSetupHardwareCursor return error.

Could you tell me how to call the IddCxMonitorSetupHardwareCursor function？

Tsche, 2020-11-09:
> IddCxMonitorSetupHardwareCursor return error.
> 
> Could you tell me how to call the IddCxMonitorSetupHardwareCursor function？

IddCxMonitorSetupHardwareCursor will return 0xC000000D (STATUS_INVALID_PARAMETER) when it is called before the monitor has arrived. Most likely you have been calling it in the wrong place. One place that works pretty well is after creating the SwapChainProcessor in AssignSwapChain. However I cannot guarantee this is ultimately the correct place.

Consider the following sample:
```cpp
void IndirectMonitorContext::AssignSwapChain(IDDCX_SWAPCHAIN SwapChain, LUID RenderAdapter, HANDLE NewFrameEvent)
{
    m_ProcessingThread.reset();

    auto Device = make_shared<Direct3DDevice>(RenderAdapter);
    if (FAILED(Device->Init()))
    {
        // It's important to delete the swap-chain if D3D initialization fails, so that the OS knows to generate a new
        // swap-chain and try again.
        WdfObjectDelete(SwapChain);
    }
    else
    {
        // Create a new swap-chain processing thread
        m_ProcessingThread.reset(new SwapChainProcessor(SwapChain, Device, NewFrameEvent, shm_obj));

        //create an event to get notified new cursor data
        HANDLE mouseEvent = CreateEventA(
            nullptr, //TODO set proper SECURITY_ATTRIBUTES
            false, 
            false,
            "arbitraryMouseEventName");;

        if (!mouseEvent)
        {
            //do error handling
        } 

        //set up cursor capabilities
        IDDCX_CURSOR_CAPS cursorInfo = {};
        cursorInfo.Size = sizeof(cursorInfo);
        cursorInfo.AlphaCursorSupport = true;
        cursorInfo.MaxX = 64; //TODO figure out correct maximum value
        cursorInfo.MaxY = 64; //TODO figure out correct maximum value
        cursorInfo.ColorXorCursorSupport = IDDCX_XOR_CURSOR_SUPPORT_NONE; //TODO play around with XOR cursors

        //prepare IddCxMonitorSetupHardwareCursor arguments
        IDARG_IN_SETUP_HWCURSOR hwCursor = {};
        hwCursor.CursorInfo = cursorInfo;
        hwCursor.hNewCursorDataAvailable = mouseEvent; //this event will be called when new cursor data is available

        NTSTATUS Status = IddCxMonitorSetupHardwareCursor(
            m_Monitor, //handle to the monitor we want to enable hardware mouse on
            &hwCursor
        );

        if (FAILED(Status))
        {
            //do error handling
        }
    }
}
```

This is the very minimum to get hardware cursor working. You should definitely call IddCxMonitorQueryHardwareCursor whenever the mouseEvent is called. This way you will get the new cursor shape which should be used to render the correct cursor.

jamaik7, 2021-04-25:
> Hi,
> 
> if I decide to handle the cursor through hardware using the function IddCxMonitorSetupHardwareCursor() and I configure it to be hidden in the surface, so completely avoiding to draw it, I'm currently facing a weird behaviour: the cursor is "invisible" as I want, except when I drag things on the surface. Infact, when dragging windows you'll see appear the mouse pointer drawn in the cursor position and, when the mouse click is released, it disappears again.
> 
> Honestly I think that the mouse cursor handler is written in the right way, I debugged the problem enough to say that I'm pretty sure that I'm doing it in the right way.
> 
> I'm wondering if this is caused by something that I'm missing in the mouse cursor handling or is it a (known?) bug with Iddcx.

Hi
when drag windows ,the poniter is appear because windows explorer draw the pointer on the window

b1on1cdog, 2022-04-06:
Hi @Tsche , i'm trying to write an IDD driver for a vm, the issue i have is that with the microsoft sample and other third parties, there's duplicated cursor when using Parsec, the sample you provided seems to help but only when i install the driver, after a reconnection through Parsec the mouse is duplicated again, if you can help me on this it would be really appreciated.

zhangwenjian, 2023-03-04:
I have same problem and the above example with IddCxMonitorSetupHardwareCursor didn't help.

zhang-xiang925, 2023-07-13:
> Hi when drag windows ,the poniter is appear because windows explorer draw the pointer on the window

Hi, what do you mean by drawing pointers on Window Explorer? How to disable Window Explorer from drawing pointers while dragging? Do you have a better way?

itsmikethetech, 2024-10-16:
Quick update that this has been addressed in the linked issue thread. Source with the fix is live, and the public build is coming shortly.

alexpacio, 2025-11-21:
@itsmikethetech can you provide the link to the source with the fix? Just for reference.