microsoft/Windows-driver-samples #539: Warnings/errors building XPSDrvSmpl with Visual Studio 2019

Nalsmoth, 2020-09-24:
I'm trying to build the XPSDrvSmpl solution and I've followed the instructions as best I can with the exception of using VS 2019 instead of 2017. I'm using Visual Studio 2019 (v4.8.03752) and Windows Driver Kit 10.0.19030.1000. After completing the instructions in the 'Build the Sample' section of the README, I get the following issues when trying to build the solution:

`12>------ Build started: Project: package (Package\package), Configuration: Release x64 ------
12>Building 'package' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
12>Stamping x64\Release\xdsmpl.inf
12>Stamping [Version] section with DriverVer=09/24/2020,13.17.4.40
12>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf : error 1297: (NTarm64.6.0) Device driver does not install on any devices, use primitive driver if this is intended.
12>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(19-19): warning 1324: [Version] section should specify PnpLockdown=1.
12>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(71-71): warning 2083: Section [printerpackageinstallation.x86] not referenced or used.
12>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(75-75): warning 2083: Section [printerpackageinstallation.amd64] not referenced or used.
12>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(79-79): warning 2083: Section [printerpackageinstallation.ia64] not referenced or used.
12>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(83-83): warning 2083: Section [printerpackageinstallation.arm64] not referenced or used.
12>Done building project "package.VcxProj" -- FAILED.
`

An older fix for the first error (which might be totally wrong?) suggested that I need to add a placeholder hardware ID in the form of a GUID to the .INF file, so I tried this in all the [Standard.*] sections:

`
[Standard.NTamd64]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_PRE_VISTA,,{D5E69F2C-BC88-406B-98E7-BF767BEABC41}
`

... and while that did get rid of the error, I now have a slightly different set of warnings and the driver still cannot be installed locally.

`
2>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(51-51): warning 1296: (NTx86,NTia64,NTamd64) Hardware 'XPSDrv Sample Driver' does not have an associated service using install section 'INSTALL_XDSMPL_FILTERS_PRE_VISTA'.
2>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(61-61): warning 1296: (NTx86.6.0,NTia64.6.0,NTamd64.6.0,NTarm64.6.0) Hardware 'XPSDrv Sample Driver' does not have an associated service using install section 'INSTALL_XDSMPL_FILTERS_VISTA'.
2>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(70-70): warning 2083: Section [printerpackageinstallation.x86] not referenced or used.
2>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(74-74): warning 2083: Section [printerpackageinstallation.amd64] not referenced or used.
2>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(78-78): warning 2083: Section [printerpackageinstallation.ia64] not referenced or used.
2>C:\...\XPSDrv_Driver_and_Filter_Sample(2)\install\xdsmpl.inf(82-82): warning 2083: Section [printerpackageinstallation.arm64] not referenced or used.
2>Building 'xdsmplcmn' with toolset 'WindowsApplicationForDrivers10.0' and the 'Desktop' target platform.
2>Building 'xdfltcmn' with toolset 'WindowsApplicationForDrivers10.0' and the 'Desktop' target platform.
2>Building 'xdsdbg' with toolset 'WindowsApplicationForDrivers10.0' and the 'Desktop' target platform.
2>Building 'xdcont' with toolset 'WindowsApplicationForDrivers10.0' and the 'Desktop' target platform.
2>........................................
2>Signability test complete.
2>
2>Errors:
2>None
2>
2>Warnings:
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntx86] of package\xdsmpl.inf
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntia64] of package\xdsmpl.inf
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntamd64] of package\xdsmpl.inf
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntx86.6.0] of package\xdsmpl.inf
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntia64.6.0] of package\xdsmpl.inf
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntamd64.6.0] of package\xdsmpl.inf
2>22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista,,{d5e69f2c-bc88-406b-98e7-bf767beabc41} in section [standard.ntarm64.6.0] of package\xdsmpl.inf
2>
`
When I try to install the printer driver, I get an error message saying "Printer driver was not installed. A problem was encountered while attempting to add the driver to the store."

There are any additional steps that need to be taken to get this sample to build on VS 2019 and install? My goal is to develop a virtual print driver to integrate with a custom port monitor, and I thought this sample would be a good starting point.

Nalsmoth, 2020-12-15:
No, I did not ever get this sample to work. But my team ended up using a different print driver altogether, so I'm not being held up by this.

AldoCrono, 2021-04-15:
Thank you @Nalsmoth , your fix for **error 1297 : "(NTarm64.6.0) Device driver does not install on any devices, use primitive driver if this is intended"** allowed me to deploy the driver on my target computer.

Also (I hope this helps someone) another step that seems to be unclear in the documentation is making sure that the model-name in your package-property-page is matching your *.inf file, as mentioned [here ](https://social.msdn.microsoft.com/Forums/sqlserver/en-US/1c8086df-e07a-4a20-85dc-2cea5c2e4189/building-a-basic-v4-printer-driver-error?forum=wdk). 

So according to the sample your model name would be:

[image]

But the thing is that there is not much information about what to do next. So would you let me know which driver did your team ended up using?

Nalsmoth, 2021-04-15:
For what it's worth, we ended up using MFILEMON, which is a custom port monitor, in order to capture raw postscript data from the printer and write it to a folder where another tool would convert it to whichever format we needed and import it into our application.

Prakash19921206, 2021-08-22:
Having same problem, project only builds if we do the following as Nalsmoth mentioned 

```
[Standard.NTamd64]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_PRE_VISTA,,{D5E69F2C-BC88-406B-98E7-BF767BEABC41}
```

have enabled test mode `bcdedit /set testsigning on`
But while installing it says "Printer driver was not installed. The system cannot find the file specified"
Using Visual studio 2019, latest Spectre libraries, Windows SDK 10.0.19041.685, Windows WDK 10.0.22000.1, Windows 10 version 20H2

anything else that's needs to be done for get this working?

Thanks & Regards
Prakash

mgaffigan, 2024-07-31:
Here's a full example with hardware ID's that bypasses `Device driver does not install on any devices, use primitive driver if this is intended`:
```inf
[Version]
Signature="$Windows NT$"
Provider=%ManufacturerName%
ClassGUID={4D36E979-E325-11CE-BFC1-08002BE10318}
Class=Printer
DriverVer=7/3/2024,18.47.35.596
CatalogFile=sample539.cat
PnpLockDown=1

[Manufacturer]
%ManufacturerName%=Standard,NTx86,NTamd64

[Standard.NTx86]
"Sample XPS Printer"=SampleXpsPrinter,{7A96A264-864F-4615-B074-5F1007B17782}

[Standard.NTamd64]
"Sample XPS Printer"=SampleXpsPrinter,{7A96A264-864F-4615-B074-5F1007B17782}

[SampleXpsPrinter]
CopyFiles=SampleXpsPrinterFiles
DriverFile=mxdwdrv.dll
ConfigFile=UniDrvUI.dll
HelpFile=UniDrv.HLP
DataFile=sample539.GPD
CoreDriverSections="{D20EA372-DD35-4950-9ED8-A6335AFE79F0},UNIDRV.OEM", "{D20EA372-DD35-4950-9ED8-A6335AFE79F5},XPSDRV.OEM,XPSGPD.OEM"
DriverCategory=PrintFax.Printer.Service

[SampleXpsPrinter.Services]
AddService=,2

[PrinterPackageInstallation.x86]
PackageAware=TRUE
CoreDriverDependencies={D20EA372-DD35-4950-9ED8-A6335AFE79F0}, {D20EA372-DD35-4950-9ED8-A6335AFE79F5}

[PrinterPackageInstallation.amd64]
PackageAware=TRUE
CoreDriverDependencies={D20EA372-DD35-4950-9ED8-A6335AFE79F0}, {D20EA372-DD35-4950-9ED8-A6335AFE79F5}

[DestinationDirs]
DefaultDestDir=66000

[SampleXpsPrinterFiles]
sample539.gpd
sample539-pipelineconfig.xml

[SourceDisksNames.x86]
1=%Location%,,

[SourceDisksNames.amd64]
1=%Location%,,

[SourceDisksFiles]
sample539.gpd                    = 1
sample539-pipelineconfig.xml     = 1

[Strings]
Location="Sample XPS Printer installation media"
ManufacturerName="Sample Manufacturer"
```

Note that `{7A96A264-864F-4615-B074-5F1007B17782}` should be replaced with a unique GUID for your driver.  It's not used anywhere, but should be unique.

closed 2025-12-17 by v-junyli