microsoft/Windows-driver-samples #554: Sysvad installation fails at "devcon.exe install ComponentizedAudioSample.inf": "There are no compatible drivers for this device"

DanijelDomazet, 2020-12-07:
I've tried to build and install the Sysvad driver using VS2019 - but failed.
After I run the command: `c:\tools\devcon.exe install tabletaudiosample.inf Root\sysvad_TabletAudioSample` I got error returned. Here is the info from `c:\windows\inf\setupapi.dev.log`: 

>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - Root\Sysvad_ComponentizedAudioSample]
>>>  Section start 2020/12/07 11:07:24.709
      cmd: c:\Tools\devcon.exe  install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample
     ndv: INF path: C:\SysvadDriver\ComponentizedAudioSample.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\MEDIA\0000}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'C:\SysvadDriver\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 11:07:24.964
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 11:07:25.008
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 11:07:25.038
     dvi:           Using exported function 'MediaClassInstaller' in module 'C:\WINDOWS\system32\mmci.dll'.
     dvi:           Class installer == mmci.dll,MediaClassInstaller
     dvi:           Class installer: Enter 11:07:25.073
     dvi:           Class installer: Exit
     dvi:           Default installer: Enter 11:07:25.091
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 11:07:25.204
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0000'.
!    ndv: No devices were updated.
     dvi: {DIF_DESTROYPRIVATEDATA} 11:07:25.276
     dvi:      Class installer: Enter 11:07:25.288
     dvi:      Class installer: Exit
     dvi:      Default installer: Enter 11:07:25.310
     dvi:      Default installer: Exit
     dvi: {DIF_DESTROYPRIVATEDATA - exit(0xe000020e)} 11:07:25.334
<<<  Section end 2020/12/07 11:07:25.345
<<<  [Exit status: FAILURE(0x00000103)]

I also installed `ComponentizedAudioSampleExtension.inf` and lastly `ComponentizedApoSample.inf`. 

All this resulted with `Unknown device` under `Other devices` in the Device Manager. 

What could be the problem?

DanijelDomazet, 2020-12-10:
The above problem happens when the sln Active Configuration is set to Debug Win32. 

When I rebuilt the projects with Release x64 configuration, it installed the drivers OK. 

Why would it not work with Debug Win32?

AbhimanyuAryan, 2020-12-27:
@DanijelDomazet have you found a solution? Even I get this error no matching drivers found `decvon failed` while installation of ComponentAudio....

DanijelDomazet, 2020-12-28:
Hi Abhimanyua, I don't know, it is behaving chaotic. Maybe I was just lucky to make it work with x64 Release. I tried another target machine, and Debug also worked.

AbhimanyuAryan, 2020-12-28:
@DanijelDomazet it worked for me on x64. But I think there's some exception which gave me BSOD. But got installed

AbhimanyuAryan, 2020-12-28:
I'll help you once I figure this out and get it to work

AbhimanyuAryan, 2021-01-02:
> The above problem happens when the sln Active Configuration is set to Debug Win32.
> 
> When I rebuilt the projects with Release x64 configuration, it installed the drivers OK.
> 
> Why would it not work with Debug Win32?

Win32 is for x86. You test machine should be 32bit x86

closed 2025-11-11 by v-junyli