microsoft/Windows-driver-samples #556: How to make sure the driver works?

DanijelDomazet, 2020-12-10:
The documentation at https://github.com/microsoft/Windows-driver-samples/tree/master/audio/sysvad under "Test the driver" says: 

"Locate an MP3 or other audio file on the target computer and double-click to play it. Then in the Sound dialog box, verify that there is activity in the volume level indicator associated with the SYSVAD (with APO Extensions) driver."

After I installed the driver, and restarted the machine, the volume level indicator blinks and stays "on", but doesn't move anymore. I hear no sound. 

When you say " activity in the volume level indicator", I suppose that also means that there is sound comming out of speakers?

DanijelDomazet, 2020-12-16:
Here is the missing info: the sound is not played back, it is saved to a file named c:\STREAM_HOST_*.wav. It's hard to believe this isn't mentioned in the documentation.

AriesSunMS, 2024-12-11:
After testing the latest code, I can not find any files like STREAM_HOST_*.wav in C:\.
Could anyone help with this issue?

guliteb, 2024-12-30:
I have the same question as @AriesSunMS . Anyone can help to clarify?

AriesSunMS, 2024-12-30:
> I have the same question as [@AriesSunMS](https://github.com/AriesSunMS) . Anyone can help to clarify?

I have resolved this issue.
1. Firstly, saving to files is turned of by default. Look for the code below in adapter.cpp, and change g_DoNotCreateDataFiles to 0.
DWORD g_DoNotCreateDataFiles = 1;  // default is off.

2. I modified the file path in savedata.cpp as below

#define DEFAULT_FILE_FOLDER1        L"\\DosDevices\\C:\\Audio_Samples"
#define DEFAULT_FILE_FOLDER2        L"\\DosDevices\\C:\\Audio_Samples\\SimpleAudioSample"
#define DEFAULT_FILE_NAME           L"\\DosDevices\\C:\\Audio_Samples\\SimpleAudioSample\\STREAM"

guliteb, 2025-01-01:
@AriesSunMS Thanks for the response. I will have a try.