microsoft/Windows-driver-samples #563: How to enable swap apo for sysvad

tomxue, 2020-12-23:
Below is Microsoft's official project to demo how to make audio driver. https://github.com/microsoft/Windows-driver-samples/tree/master/audio/sysvad

After making build, 3 APO dll files are generated. And those inf files contain the installation steps for them. But I tested and found that those apo effects do not take effect at all. My test method is: I empty all the APOProcess() in those 3 APO and found that audio recorded is not influenced.

Question 1： My understanding is that these APO dlls must be invoked by some programs and then we could have sound effects. While sysvad does not provide such programs. Am I right？

Question 2： I want to know how do those APO codes interact with sysvad microphone driver? By code, or by some id in the inf file?

Question 3: Furthermore, if I want to reuse the APO codes for other microphone device driver, is it possible and how to modify them? E.g. for my realtek device driver which I have no source code.

Thanks!

AbhimanyuAryan, 2021-01-17:
@tomxue has Microsoft explained sysvad project/code/files? Also, you mentioned APO sound effects. What do you mean by that?

naveenkunni, 2021-07-29:
Hi @tomxue , 

Did you get any response or make any progress on this? I'm stuck at the same point, all I could find was that the Sysvad driver and Swap/delay/kws APO have been migrated to the Universal driver framework, and so the App interface (the Properties dialog) has been removed.

If you were able to make progress, please let me know.

Regards,
Naveen

lawlict, 2022-01-07:
Hi @tomxue, Have you solved the problem? I'm also looking for the answer. Thanks.