microsoft/Windows-driver-samples #565: Sysvad installation fails for Debug x64 for ComponentizedAudioSample.inf

AbhimanyuAryan, 2020-12-28:
System configurations for Development Machine(PC) and Target Machine(Macbook Bootcamp)

* Windows 10 2004 19041.685
* Visual Studio 2019 community 16.8.3
* Windows 10 SDK (10.0.19041.0)
* Windows Driver Kit

So basically both machines have same setup

The complication on both machines is happening just fine. Creating package file. Instead of win32 debug, I used x64 Debug in configuration manager

Also, `devcon.exe` is also from `x64` directory

I have installed certificate on target machine from `C:/Certificates/package.cer`  which was moved from Development Machine 

kept package files in target machine in `C:/SysvardDriver`
- ComponentizedApoSample.inf
- ComponentizedAudioSample.inf
- ComponentizedAudioSampleExtension.inf
- DelayAPO.dll
- KeywordDetectorContosoAdapter.dll
- KWSApo.dll
- SwapAPO.dll
- sysvad.cat
- tabletaudiosample.inf
- TabletAudioSample.sys

Error in CMD:

```
C:\SysvadDriver>C:\Tools\devcon install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample
Device node created. Install is complete when drivers are installed...
Updating drivers for Root\Sysvad_ComponentizedAudioSample from C:\SysvadDriver\ComponentizedAudioSample.inf.
devcon failed.
```

Detailed INF error from `Windows/INF/setupapi.dev`

```
>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - Root\Sysvad_ComponentizedAudioSample]
>>>  Section start 2020/12/28 08:41:16.300
      cmd: C:\Tools\devcon  install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample
     ndv: INF path: C:\SysvadDriver\ComponentizedAudioSample.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\MEDIA\0000}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'C:\SysvadDriver\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 08:41:16.346
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 08:41:16.366
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 08:41:16.376
     dvi:           Default installer: Enter 08:41:16.382
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 08:41:16.412
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0000'.
     ndv: {Update Device Driver - ROOT\MEDIA\0001}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'C:\SysvadDriver\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 08:41:16.452
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 08:41:16.469
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 08:41:16.479
     dvi:           Default installer: Enter 08:41:16.485
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 08:41:16.514
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0001'.
!    ndv: No devices were updated.
<<<  Section end 2020/12/28 08:41:16.545
<<<  [Exit status: FAILURE(0x00000103)]
```

AbhimanyuAryan, 2020-12-28:
also I had some issues while compiling because timestamp something. So I used `Properties > INF2CAT > /uselocaltime`
after that project complied

AbhimanyuAryan, 2020-12-28:
If I right click and install `ComponentizedAudioSample.inf`. I get this prompt. I have already installed the `package.cer` to local as well and user
[image]

AbhimanyuAryan, 2020-12-28:
[image]

unknown device as mentioned in this issue by Dan: https://github.com/microsoft/Windows-driver-samples/issues/554

mvaneerde, 2020-12-28:
After a private chat with Abi we figured this out.
The driver is being compiled for x86, and installed on amd64.

closed 2020-12-28 by AbhimanyuAryan