microsoft/Windows-driver-samples #567: cannot Debug the Virtual serial port Driver for windows

InvBoy01001, 2021-01-09:
this is my first time writing driver
I found the virtual serial port from [https://docs.microsoft.com/en-us/samples/microsoft/windows-driver-samples/virtual-serial-driver-sample/](url) and I tried to run it but it's get this error ( I didn't change the code )
`Spectre-mitigated libraries are required for this project.
Install them from the Visual Studio installer (Individual components tab)
for any toolsets and architectures being used.
Learn more: https://aka.ms/Ofhn4c   fakemodem (FakeModem\fakemodem)
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets`
what should I do?

zodiacon, 2021-01-09:
You can go back to Visual Studio installer, and add the Spectre-mitigation libraries to your installation.
Alternatively, you can change project properties to use the non-Spectre libraries

InvBoy01001, 2021-01-09:
> You can go back to Visual Studio installer, and add the Spectre-mitigation libraries to your installation.
> Alternatively, you can change project properties to use the non-Spectre libraries

I install all of them?

zodiacon, 2021-01-09:
Just the latest version with the CPU arch you intend to compile

InvBoy01001, 2021-01-09:
> Just the latest version with the CPU arch you intend to compile

there is a lot of them

Compilers, build tools, and runtimes

MSVC v142 - VS 2019 C++ ARM Spectre-mitigated libs (v14.28)

MSVC v142 - VS 2019 C++ ARM64 Spectre-mitigated libs (v14.28)

MSVC v142 - VS 2019 C++ x64/x86 Spectre-mitigated libs (v14.28)

SDKs, libraries, and frameworks

C++ ATL for latest v142 build tools with Spectre Mitigations (ARM)

C++ ATL for latest v142 build tools with Spectre Mitigations (ARM64)

C++ ATL for latest v142 build tools with Spectre Mitigations (x86 & x64)

C++ ATL for v141 build tools with Spectre Mitigations (ARM)

C++ ATL for v141 build tools with Spectre Mitigations (ARM64)

C++ ATL for v141 build tools with Spectre Mitigations (x86 & x64)

C++ MFC for latest v142 build tools with Spectre Mitigations (ARM)

C++ MFC for latest v142 build tools with Spectre Mitigations (ARM64)

C++ MFC for latest v142 build tools with Spectre Mitigations (x86 & x64)

C++ MFC for v141 build tools with Spectre Mitigations (ARM)

C++ MFC for v141 build tools with Spectre Mitigations (ARM64)

C++ MFC for v141 build tools with Spectre Mitigations (x86 & x64)

C++ v14.20 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.20 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.20 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.20 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.20 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.20 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.21 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.21 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.21 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.21 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.21 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.21 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.22 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.22 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.22 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.22 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.22 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.22 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.23 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.23 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.23 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.23 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.23 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.23 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.24 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.24 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.24 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.24 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.24 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.24 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.25 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.25 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.25 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.25 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.25 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.25 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.26 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.26 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.26 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.26 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.26 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.26 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.27 ATL for v142 build tools with Spectre Mitigations (ARM)

C++ v14.27 ATL for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.27 ATL for v142 build tools with Spectre Mitigations (x86 & x64)

C++ v14.27 MFC for v142 build tools with Spectre Mitigations (ARM)

C++ v14.27 MFC for v142 build tools with Spectre Mitigations (ARM64)

C++ v14.27 MFC for v142 build tools with Spectre Mitigations (x86 & x64)

zodiacon, 2021-01-09:
These should suffice:
MSVC v142 - VS 2019 C++ ARM Spectre-mitigated libs (v14.28)
MSVC v142 - VS 2019 C++ ARM64 Spectre-mitigated libs (v14.28)
MSVC v142 - VS 2019 C++ x64/x86 Spectre-mitigated libs (v14.28)

AddioElectronics, 2021-11-07:
Mind if I high jack this thread for a second?

I have compiled the same driver, and have it installed on a USB device. (Samd21J Xplained Pro)

When I try to connect to from a serial terminal it fails to connect.

So I made a quick little C# serial terminal, and when I call `SerialPort.Open` I get `System.ArgumentException,`
with this error message.
 `{"The given port name is invalid.  It may be a valid port, but not a serial port.\r\nParameter name: portName"}`

Any ideas?