microsoft/Windows-driver-samples #570: Windows-driver-samples/video/IndirectDisplay/IddSampleDriver/Driver.cpp has a resource leak

mr-intj, 2021-01-14:
As far as I can tell, the `Cleanup()` method for the `IndirectMonitorContextWrapper` struct is never called.
```
struct IndirectMonitorContextWrapper
{
    IndirectMonitorContext* pContext;

    void Cleanup()
    {
        delete pContext;
        pContext = nullptr;
    }
};
```
All `IndirectMonitorContext` instances allocated with `new` are never destroyed.

jennaric, 2021-07-29:
hello,do you see iddcxadapterinitasync function returned error code 0xc00000bb STATUS_ NOT_ Supported, thanks

mr-intj, 2021-07-30:
> hello,do you see iddcxadapterinitasync function returned error code 0xc00000bb STATUS_ NOT_ Supported, thanks

Not sure whether you are referring to my post, or asking a new question. In my case, I'm saying that even when there are no errors, this `Cleanup` method never gets called.

jennaric, 2021-08-02:
Hello, I mean when you test iddCXAdapterInitAsync return value is normal SUCCESS or FAILED

mr-intj, 2021-08-03:
OK. I'm not getting an error back from `::IddCxAdapterInitAsync`:
```
    ////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
    // Start the initialization of the IddCx adapter, which will trigger a call to CAdapter::OnIddCxAdapterInitFinished after this
    // method returns.

    IDARG_OUT_ADAPTER_INIT initOut{};
    auto ntStatus = ::IddCxAdapterInitAsync(&initIn, &initOut);
    if (!NT_SUCCESS(ntStatus))
        TraceError("IddCxAdapterInitAsync failed; %!STATUS!", ntStatus);

```

jennaric, 2021-08-03:
Got it. Thank you. which win10 version with you used

closed 2025-11-11 by v-junyli