microsoft/Windows-driver-samples #573: .inf not allowing me to compile hello world WDM empty driver

codingonsteroid, 2021-01-22:
building my IO driver. INF file is not letting me compile it

```
;
; FifyDriver.inf
;

[Version]
Signature="$WINDOWS NT$"
Class=System
ClassGuid={4d36e97d-e325-11ce-bfc1-08002be10318}
Provider=%ManufacturerName%
DriverVer=
CatalogFile=FifyDriver.cat
PnpLockDown=1

[DestinationDirs]
DefaultDestDir = 12

[SourceDisksNames]
1 = %DiskName%,,,""

[SourceDisksFiles]

[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$

[Standard.NT$ARCH$]

[Strings]
ManufacturerName="<Your manufacturer name>" ;TODO: Replace with your manufacturer name
ClassName=""
DiskName="FifyDriver Source Disk"
```

I get this error: 
```Severity    Code    Description    Project    File    Line    Suppression State
Error    1297    Device driver does not install on any devices, use primitive driver if this is intended.    FifyDriver    C:\Users\testpc\source\repos\FifyDriver\FifyDriver\FifyDriver.inf    1
```

If I remove based on Stackoverflow answer: https://stackoverflow.com/questions/64197518/device-driver-does-not-install-on-any-devices-use-primitive-driver-if-this-is-i
```
[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$
```

then I get some other error

```
Inf2Cat Tool Output:
............
Signability test failed.

Errors:
No installation INF found in the root path of the driver. For the driver to be digitally signed, and for it to install properly the installation inf must be in the root of each driver path you have provided. If you have separate folders with different driver packages for different languages, operating systems or device categories then specify each driver set as a separate driver. Each driver package must be selected separately and must contain an installation INF.

Warnings:
None
```

I remove .inf from solution explorer and now able to build .sys from c code

[image]

vtorres, 2021-05-19:
From:
```
[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$
```

To:
```
[Manufacturer]
```

Keeping the [Manufactorer], without information below it

vtorres, 2021-05-19:
From:
```
[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$
```
To:
```
[Manufacturer]
```
Keeping the [Manufacturer], without information below it

vtorres, 2021-05-19:
You probably removed all the manufacturer line, so you must do this:

From:
```
[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$
```
To:
```
[Manufacturer]
```

Keeping the [Manufacturer], without information below it

taianrc, 2022-11-17:
> 您可能删除了所有制造商行，因此您必须这样做：
> 
> 从：
> 
> ```
> [Manufacturer]
> %ManufacturerName%=Standard,NT$ARCH$
> ```
> 
> 至：
> 
> ```
> [Manufacturer]
> ```
> 
> 保留[制造商]，下面没有信息
 install this driver but failed

[image]

bharavi15, 2023-12-13:
@vtorres This works! Thank you 👍🏻

closed 2025-11-11 by v-junyli