microsoft/Windows-driver-samples #583: [SimpleMediaSource sample] : Too high CPU usage

fuxupeng, 2021-02-26:
When driver has been insatalled , I open windows camera APP to preview video, it will cause Too high CPU usage. 
Question 1: Why and what can i do to make it better?
Question 2: Why is there 2 devices named "SimpleMediaSource Capture Source" ?
[image]

AntumArk, 2021-12-02:
How were you able to even run this device? 
I am either getting `Adding the driver package failed : A problem was encountered while attempting to add the driver to the store.`
 or `Adding the driver package failed : The third-party INF does not contain digital signature information.`

pmaciejew, 2022-09-16:
I also see a high cpu usage, and two camera devices. 

In order to install the driver you need to disable driver signing check. For example run "bcdedit /set bootmenu policy". Then, durring windows boot push F8 and select option to disable driver sign check.

closed 2025-11-11 by v-junyli