microsoft/Windows-driver-samples #592: Failure loading SimpleAudioSample driver - driver entry point not found (Code 39)

belargo, 2021-03-19:
**SimpleAudioSample** driver has been built with the latest version of VisualStudio 2019 (16.9.2) **for** x64 platform.
Installing driver with devcon on target machine doesn't produce any errors.
Checking installed driver status in Device Manager reveals that the driver was not loaded.
Device status:
"Windows cannot load the device driver for this hardware. The driver may be corrupted or missing (Code 39)

{Driver Entry Point Not Found}
...
See attached screenshot.
[image]

belargo, 2021-03-29:
The problem mysteriously disappeared after most recent Windows update on March 29, 2021.

closed 2021-03-29 by belargo

Thomsywj, 2021-04-25:
Please, How can I solved the issue？thx

Thomsywj, 2021-04-25:
Please，are there some updates？

IM-Boting, 2021-08-10:
> Please，are there some updates？

update to 2004 or higher. Versions above 2004 can be used normally. But I want to know how to be compatible with the lower version。

AndreiHarasemiuc, 2022-06-19:
Having this exact problem too and I'm on the most recent Windows 10 version, well beyond 2004.

Any help please?

AndreiHarasemiuc, 2022-06-19:
Can you reopen this @belargo ?