microsoft/Windows-driver-samples #600: Cannot register Mft driver dll(error code 0x8070715)

gioviopantat, 2021-04-01:
I put SampleMft0.dll under C:\Windows\System32, when I was trying to use regsvr32.exe under C:\Windows\System32 to register SampleMft0.dll but always getting error message "The module SampleMft0.dll was loaded but the call to DllRegisterServer failed with error code 0x80070715".

(I've also put SampleMft0.dll under C:\Windows\System32 and C:\Windows\SysWOW64, it still pop out the same error message)

Here's debug view msg:

C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VS\Tools\MSVC\14.28.29910\altmfc\include\statreg.h : altTraceRegistar - Failed to FindResource on ID:101 TYPE:REGISTRY

Btw don't know if it's matter, but I already add CLSID {7BB640D9-33A4-4759-B290-F41A31DCF848} into regedit.

---------------------------------------------------------------------------------------------------------------------------------------------

OS version: Windows 10 Professional 18363.1440

VS version: VS 2019

Build env: Debug x64

jmxntg, 2021-04-01:
Dear @gioviopantat 

I've adjusted some of the project parameters and now it is generating .dll files that can be registered without error.
Build the project instead of building the entire solution.

Please let me know if this works for you.

[samplemft0.zip](https://github.com/microsoft/Windows-driver-samples/files/6243884/samplemft0.zip)

gioviopantat, 2021-04-06:
Hi @jmxntg 

 Appreciate your help, but I still got error code 0x8070715 while registering the dll.

Thank you

gioviopantat, 2021-04-06:
Hi @jmxntg 

Appreciate your help, I'll update result very soon.

BR

n20, 2021-08-30:
@jmxntg I have downloaded the zip file you posted here and built the project, then tried to register the DLL, and I have the same problem still. @gioviopantat have you been able to get the DLL to register succesfully? I also tried the install_MS_mft0.bat batch script to install the DLL, same error code. I have tried both 32-bit and 64-bit versions, debug and release. I am calling regsvr32.exe from a VS elevated (admin) cmd prompt and the DLL has been copied to \program files or \program files (x86) respectively. 

windows 10 version 20H2 (OS build 19042.1165) 
Microsoft Visual Studio Community 2019 Version 16.10.3

n20, 2021-08-30:
I copied my x64 release build of SampleMft0 (compiled from your above code) into C:\Program Files\SampleMft0 where I run regsvr32.exe (still same error code). I have used dumpbin to figure out the dependencies. 

C:\Program Files\SampleMft0>dumpbin /dependents SampleMft0.dll
Microsoft (R) COFF/PE Dumper Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file SampleMft0.dll

File Type: DLL

  Image has the following dependencies:

    KERNEL32.dll
    USER32.dll
    ADVAPI32.dll
    ole32.dll
    OLEAUT32.dll
    MFPlat.DLL
    api-ms-win-core-winrt-string-l1-1-0.dll

  Summary

        1000 .00cfg
        4000 .data
        2000 .idata
        6000 .pdata
       28000 .rdata
        2000 .reloc
        1000 .rsrc
       7A000 .text
        1000 _RDATA

C:\Program Files\SampleMft0>

the above DLLs are all in C:\windows\syswow64, with the exception of api-ms-win-core-winrt-string-l1-1-0.dll
which seems to be located on my machine in C:\Program Files (x86)\Microsoft SDKs\NuGetPackages\Microsoft.NETCore.Windows.ApiSets-x86\1.0.0\runtimes\win7-x86\native

I tried copying this DLL into c:\windows\syswow64 and tried regsvr32.exe again, but still the same error message as before. The other DLL files are already in c:\windows\syswow64.

n20, 2021-08-30:
FWIW, I do not have anything in this folder - in C:\Program Files (x86)\Microsoft SDKs\NuGetPackages\Microsoft.NETCore.Windows.ApiSets-x86\1.0.0\runtimes\win10-x86\native