microsoft/Windows-driver-samples #657: Version compatibility problem: Win10 earlier version failed to install the driver, code 39

IM-Boting, 2021-08-04:
High version of WDK written driver will be incompatible with low version of Win10? I followed the steps to install the driver development environment, configure the SimpleAudioSample project and generate it. I then install the generated file in device Manager, prompting :Windows cannot load the driver for this hardware device, the driver may be corrupted or missing,(code 39). Magic is that my colleagues can correctly install, after a day of investigation I suspected to be Win10 version problems. But see Microsoft document inside write support win7 above version, here I produced doubt, hope to get a solution.
Project compilation environment :VS2019 Win10SDk10.0.19041.0 WDK10.0.19041.0
My system: Win10 1909(OS Internal version 18363.592)
Win10 2004(OS Internal 19041.388)
<---------------------------------------------------------------------------------------->
高版本WDK编写的的驱动程序是否会不兼容低版本的win10？我按照步骤安装驱动开发环境，配置SimpleAudioSample项目并生成。然后我在设备管理器中安装生成的文件，提示：Windows 无法加载这个硬件设备的驱动程序 驱动程序可能已损坏或不见了(代码39)。神奇的是我的同事可以正确安装，经过一天的排查后怀疑是win10版本问题，但看见微软文档里面写的支持win7以上的版本。这里我产生了疑惑，希望得到解答。
项目编译环境:VS2019 win10SDK10.0.19041.0 WDK10.0.19041.0
本人系统:win10 1909(OS内部版本 18363.592)
同事系统:win10 2004(OS内部版本 19041.388)

MorphKyan, 2021-08-09:
Same question

Guawazii, 2021-08-13:
Did you solve this problem? @MorphKyan

IM-Boting, 2021-09-13:
> Same question

Hi,This may give you some help。
--------------------------------------------------------------
我的问题是2004及以上的版本使用了ExAllocatePool2，低版本不执行，要换成之前的函数。我直接使用了ExAllocatePoolWithTag。除了替换函数，还要进行的操作是把POOL_FLAGS类型的参数换成POOL_TYPE。
https://docs.microsoft.com/zh-cn/windows-hardware/drivers/kernel/updating-deprecated-exallocatepool-calls

IM-Boting, 2021-09-13:
> Did you solve this problem? @MorphKyan
https://docs.microsoft.com/zh-cn/windows-hardware/drivers/kernel/updating-deprecated-exallocatepool-calls

MorphKyan, 2021-09-15:
@IM-Boting 
Thanks, and I have found another repository that might be useful. @Guawazii 
https://github.com/JannesP/AudioMirror

IM-Boting, 2021-09-16:
> @IM-Boting
> Thanks, and I have found another repository that might be useful. @Guawazii
> https://github.com/JannesP/AudioMirror

I have a question about this project. It does not appear to use `IRP MJ DEVICE CONTROL`, So how to write PCM data to the virtual microphone.🤔

MorphKyan, 2021-09-16:
> > @IM-Boting
> > Thanks, and I have found another repository that might be useful. @Guawazii
> > https://github.com/JannesP/AudioMirror
> 
> I have a question about this project. It does not appear to use `IRP MJ DEVICE CONTROL`, So how to write PCM data to the virtual microphone.🤔

Sorry, i just use it as a virtual device. I don't know it well. :(

xinlin-z, 2022-09-29:
> > Same question
> 
> ## Hi,This may give you some help。
> 
> 我的问题是2004及以上的版本使用了ExAllocatePool2，低版本不执行，要换成之前的函数。我直接使用了ExAllocatePoolWithTag。除了替换函数，还要进行的操作是把POOL_FLAGS类型的参数换成POOL_TYPE。 https://docs.microsoft.com/zh-cn/windows-hardware/drivers/kernel/updating-deprecated-exallocatepool-calls

May I ask if this problem is sovled by the solution you post above?

closed 2025-11-11 by v-junyli