microsoft/Windows-driver-samples #659: Can not run EXE project

volehuy1998, 2021-08-24:
VS2019 Windows Filtering Platform **EXE project**
I turned on testsigning and debug. But it didn't work.
https://user-images.githubusercontent.com/36606761/130547916-e65c529e-c07d-404a-bc75-21b77012a282.png

volehuy1998, 2021-08-24:
[image]

tqk2811, 2021-09-29:
Me too, reason here  
[image]
-----------------------------
I was try turn off [Secure Boot](https://answers.microsoft.com/en-us/protect/forum/all/your-organization-used-windows-defender/ef12b7be-41f5-4b71-a73c-e4f99ba944f7) but it not work

volehuy1998, 2021-09-29:
@tqk2811 I have solved it, just missing something about strong digital signature. Let try with me.
We will use some already tool in this folder: makecert.exe, pvk2pfx.exe and signtool.exe
`
cd "C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x64\"
`
After that, run this command
```
makecert.exe -sv MyApp.pvk -n "CN=MyCompany" -len 2048 MyApp.cer -r
pvk2pfx.exe -pvk MyApp.pvk -spc MyApp.cer -pfx MyApp.pfx -po MyPassword
signtool.exe sign /f MyApp.pfx /p MyPassword "Release\WFPSampler.exe"
```
Use signtool for all all of WFPSampler solution output (driver, service, ...).

After sign all output, I've deployed and test on virtual machine (bridge static ip192.168.1.62), it worked very well. Below image that WFPSampler blocked packet come from Google DNS server (8.8.8.8)
[image]

closed 2021-09-29 by volehuy1998