microsoft/Windows-driver-samples #660: sampledevicemft not working? no luck with mftrace

n20, 2021-08-28:
Hi all

(I also posted this question at https://docs.microsoft.com/en-us/answers/questions/531553/sampledevicemft-not-working-no-luck-with-mftrace.html) 

I'm trying to build a device MFT and I'm using this sample
https://github.com/Microsoft/Windows-driver-samples/tree/master/avstream/sampledevicemft

The sample does not build out of the box, so I fixed the missing include for WIL - ./packages/Microsoft.Windows.ImplementationLibrary.1.0.210204.1/include/

I then was able to build the sample and register the DLL using regsvr32.exe. 
I then started the camera app in windows 10 hoping I would see some kind of effect applied to the live video preview stream, but the video stream just looks normal.

I then enabled the MF_DEVICEMFT_ADD_GRAYSCALER_ define in stdafx.h thinking this was what was missing, but when I do that, I get a bunch of compilation errors. I tried my best to fix them and managed to build the code again and register the DLL, but still, nothing has changed in the video output when I start the camera app, so I thought I'd trace with mftrace, since the sample is using WPP for tracing? 
However, that doesn't work either. here is my config.xml file - 

<?xml version='1.0' encoding='utf-8'?>
<providers>
  <mfdetours level="Debug"> 
    <keyword ID="All" />
  </mfdetours>
  
  

  <provider level="verbose" ID="{0E313280-3169-4F41-A329-9E854169634F}">
    <keyword ID="All" />
  </provider>

</providers>

Any ideas on how I can get the sample working and how to verify the DLL is actually being loaded and used when the video stream of my camera is opened? 

In the registry I manually added a string under this node - 
Computer\HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Control\DeviceClasses\{e5323777-f976-4f5b-9b55-b94699c46e44}\##?#USB#VID_046D&PID_0821&MI_02#7&4c057cf&0&0002#{e5323777-f976-4f5b-9b55-b94699c46e44}\#GLOBAL\Device Parameters

name = CameraDeviceMftClsid
data = {0E313280-3169-4F41-A329-9E854169634F}

I am building in release, 64-bit. 

Thanks
B

n20, 2021-08-28:
<?xml version='1.0' encoding='utf-8'?>
<providers>
  <mfdetours level="Debug"> 
    <keyword ID="All" />
  </mfdetours>
  
  

  <provider level="verbose" ID="{0E313280-3169-4F41-A329-9E854169634F}">
    <keyword ID="All" />
  </provider>

</providers>

config.xml for mftrace

n20, 2021-08-28:
`<?xml version='1.0' encoding='utf-8'?>
<providers>
  <mfdetours level="Debug"> 
    <keyword ID="All" />
  </mfdetours>
  
  

  <provider level="verbose" ID="{0E313280-3169-4F41-A329-9E854169634F}">
    <keyword ID="All" />
  </provider>

</providers>
`

n20, 2021-08-28:
I can't post the config.xml I use with mftrace, see the link here to find it - 
https://docs.microsoft.com/en-us/answers/questions/531553/sampledevicemft-not-working-no-luck-with-mftrace.html

jiteshkris, 2021-08-31:
Can you please post the camera traces here?
aka.ms/cameratrace

n20, 2021-08-31:
@jiteshkris when I follow that link and sign in I get this error message - AADSTS50020: User account '...' from identity provider 'live.com' does not exist in tenant 'Microsoft' and cannot access the application '44218cb0-1b0e-441c-9092-0dce257e9eb9'(osgwikiazureprod) in that tenant. The account needs to be added as an external user in the tenant first. Sign out and sign in again with a different Azure Active Directory user account.

Sudhaushu, 2021-09-01:
The best way to share the trace logs for investigation is the feedback hub.
You need to launch feedback hub on your repro machine (shortkey: Win + F   or type feedback hub in search bar).
There you will need to write
1. A title and brief description.
2. choose category , in this case - "Devices and Drivers" -> "Device Camera or Webcams"
3. Don’t choose any similar feedback, for this case
4. In the "Add more details" section - click "recreate my problem->Start recording" 
5. While the recording is going on. open the camera app (while the DMFT is installed/registered) , proceed with repro steps 
6. Stop recording and Submit the feedback.

n20, 2021-09-01:
@Sudhaushu ok, done