microsoft/Windows-driver-samples #661: WFPSampler: Your organization used windows defender application control to block this app

tqk2811, 2021-09-29:
[image]

WDK: 10.0.22000.1 and 10.0.19041.0
Win10: 21h1 build 19043.1237

<details>
<summary>Script  run result</summary>

> D:\temp\WFP>WFPSamplerInstall D:\temp\WFP x86 Win8.1 Debug
> 
> Installing WFPSampler
> 
> Attempting to sign WFPSampler.Exe
> 
> The following certificates have been found to be suitable for signing:
>     Issued to: WDKTestCert tqk28,132772986710156745
>     Issued by: WDKTestCert tqk28,132772986710156745
>     Expires:   Sun Sep 28 07:00:00 2031
>     SHA1 hash: DF6CB32FC729FC2DBC00C603180C6108B5C51FF2
> 
>     Issued to: WDKTestCert tqk28,132772076835745501
>     Issued by: WDKTestCert tqk28,132772076835745501
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 92261027B75703C82EFD4CD95C4864877D5DBF1A
> 
>     Issued to: WDKTestCert tqk28,132772075642505488
>     Issued by: WDKTestCert tqk28,132772075642505488
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 78C7B5D4D36BD08AC791BA00B3CDFBDF944495E1
> 
>     Issued to: 1cc600aa-73cc-4111-8c8a-e002e011c397
>     Issued by: 1cc600aa-73cc-4111-8c8a-e002e011c397
>     Expires:   Tue Sep 06 19:38:42 2022
>     SHA1 hash: 75DEA48DF7B20488BDC4BD92D4F97DEE33728FE4
> 
>     Issued to: WDKTestCert tqk28,132772043084978182
>     Issued by: WDKTestCert tqk28,132772043084978182
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 6DA3148768F748F196EDB6398AE48A735CEF9B37
> 
>     Issued to: WDKTestCert tqk28,132772051360482755
>     Issued by: WDKTestCert tqk28,132772051360482755
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 56F6F685CD3737DB774A1E8BADD1358F9669EE6D
> 
>     Issued to: 3ffab630-822c-4b94-96d1-5e14aeaefc30
>     Issued by: 3ffab630-822c-4b94-96d1-5e14aeaefc30
>     Expires:   Thu Sep 29 10:39:50 2022
>     SHA1 hash: 431046F2AD479254321919C5E717D23192721482
> 
> The following certificate was selected:
>     Issued to: WDKTestCert tqk28,132772986710156745
>     Issued by: WDKTestCert tqk28,132772986710156745
>     Expires:   Sun Sep 28 07:00:00 2031
>     SHA1 hash: DF6CB32FC729FC2DBC00C603180C6108B5C51FF2
> 
> Done Adding Additional Store
> Successfully signed: D:\temp\WFP\WFPSampler.exe
> 
> Number of files successfully Signed: 1
> Number of warnings: 0
> Number of errors: 0
> 
> Attempting to sign WFPSamplerService.Exe
> 
> The following certificates have been found to be suitable for signing:
>     Issued to: WDKTestCert tqk28,132772986710156745
>     Issued by: WDKTestCert tqk28,132772986710156745
>     Expires:   Sun Sep 28 07:00:00 2031
>     SHA1 hash: DF6CB32FC729FC2DBC00C603180C6108B5C51FF2
> 
>     Issued to: WDKTestCert tqk28,132772076835745501
>     Issued by: WDKTestCert tqk28,132772076835745501
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 92261027B75703C82EFD4CD95C4864877D5DBF1A
> 
>     Issued to: WDKTestCert tqk28,132772075642505488
>     Issued by: WDKTestCert tqk28,132772075642505488
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 78C7B5D4D36BD08AC791BA00B3CDFBDF944495E1
> 
>     Issued to: 1cc600aa-73cc-4111-8c8a-e002e011c397
>     Issued by: 1cc600aa-73cc-4111-8c8a-e002e011c397
>     Expires:   Tue Sep 06 19:38:42 2022
>     SHA1 hash: 75DEA48DF7B20488BDC4BD92D4F97DEE33728FE4
> 
>     Issued to: WDKTestCert tqk28,132772043084978182
>     Issued by: WDKTestCert tqk28,132772043084978182
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 6DA3148768F748F196EDB6398AE48A735CEF9B37
> 
>     Issued to: WDKTestCert tqk28,132772051360482755
>     Issued by: WDKTestCert tqk28,132772051360482755
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 56F6F685CD3737DB774A1E8BADD1358F9669EE6D
> 
>     Issued to: 3ffab630-822c-4b94-96d1-5e14aeaefc30
>     Issued by: 3ffab630-822c-4b94-96d1-5e14aeaefc30
>     Expires:   Thu Sep 29 10:39:50 2022
>     SHA1 hash: 431046F2AD479254321919C5E717D23192721482
> 
> The following certificate was selected:
>     Issued to: WDKTestCert tqk28,132772986710156745
>     Issued by: WDKTestCert tqk28,132772986710156745
>     Expires:   Sun Sep 28 07:00:00 2031
>     SHA1 hash: DF6CB32FC729FC2DBC00C603180C6108B5C51FF2
> 
> Done Adding Additional Store
> Successfully signed: D:\temp\WFP\WFPSamplerService.exe
> 
> Number of files successfully Signed: 1
> Number of warnings: 0
> Number of errors: 0
> 
> Attempting to sign WFPSamplerCaloutDriver.Sys
> 
> The following certificates have been found to be suitable for signing:
>     Issued to: WDKTestCert tqk28,132772986710156745
>     Issued by: WDKTestCert tqk28,132772986710156745
>     Expires:   Sun Sep 28 07:00:00 2031
>     SHA1 hash: DF6CB32FC729FC2DBC00C603180C6108B5C51FF2
> 
>     Issued to: WDKTestCert tqk28,132772076835745501
>     Issued by: WDKTestCert tqk28,132772076835745501
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 92261027B75703C82EFD4CD95C4864877D5DBF1A
> 
>     Issued to: WDKTestCert tqk28,132772075642505488
>     Issued by: WDKTestCert tqk28,132772075642505488
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 78C7B5D4D36BD08AC791BA00B3CDFBDF944495E1
> 
>     Issued to: 1cc600aa-73cc-4111-8c8a-e002e011c397
>     Issued by: 1cc600aa-73cc-4111-8c8a-e002e011c397
>     Expires:   Tue Sep 06 19:38:42 2022
>     SHA1 hash: 75DEA48DF7B20488BDC4BD92D4F97DEE33728FE4
> 
>     Issued to: WDKTestCert tqk28,132772043084978182
>     Issued by: WDKTestCert tqk28,132772043084978182
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 6DA3148768F748F196EDB6398AE48A735CEF9B37
> 
>     Issued to: WDKTestCert tqk28,132772051360482755
>     Issued by: WDKTestCert tqk28,132772051360482755
>     Expires:   Sat Sep 27 07:00:00 2031
>     SHA1 hash: 56F6F685CD3737DB774A1E8BADD1358F9669EE6D
> 
>     Issued to: 3ffab630-822c-4b94-96d1-5e14aeaefc30
>     Issued by: 3ffab630-822c-4b94-96d1-5e14aeaefc30
>     Expires:   Thu Sep 29 10:39:50 2022
>     SHA1 hash: 431046F2AD479254321919C5E717D23192721482
> 
> The following certificate was selected:
>     Issued to: WDKTestCert tqk28,132772986710156745
>     Issued by: WDKTestCert tqk28,132772986710156745
>     Expires:   Sun Sep 28 07:00:00 2031
>     SHA1 hash: DF6CB32FC729FC2DBC00C603180C6108B5C51FF2
> 
> Done Adding Additional Store
> Successfully signed: D:\temp\WFP\WFPSamplerCalloutDriver.sys
> 
> Number of files successfully Signed: 1
> Number of warnings: 0
> Number of errors: 0
> 
> Copying WFPSamplerCalloutDriver Bins to C:\Windows\System32\Drivers\
> D:\temp\WFP\WFPSamplerCalloutDriver.inf
> D:\temp\WFP\WFPSamplerCalloutDriver.sys
>         2 file(s) copied.
> 
> Copying WFPSampler application binaries to C:\Windows\System32\
>         1 file(s) copied.
>         1 file(s) copied.
> 
> Registering the WFPSampler Service
> The system cannot execute the specified program.
> The service name is invalid.
> 
> More help is available by typing NET HELPMSG 2185.
> 
> 
> Registering the WFPSampler Callout Driver
> The service name is invalid.
> 
> More help is available by typing NET HELPMSG 2185.

</details>

I was try disable [Secure Boot](https://answers.microsoft.com/en-us/protect/forum/all/your-organization-used-windows-defender/ef12b7be-41f5-4b71-a73c-e4f99ba944f7) but it not work.

volehuy1998, 2021-09-29:
@tqk2811 I have solved it, just missing something about strong digital signature. Let try with me (https://github.com/microsoft/Windows-driver-samples/issues/659#issuecomment-929835786)

After below command (as Administrator) please restart virtual machine:
```
bcdedit /debug on
bcdedit /set testsigning on
```
We will use some already tool in this folder: makecert.exe, pvk2pfx.exe and signtool.exe
`
cd "C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x64\"
`

After that, run this command:
```
makecert.exe -sv MyApp.pvk -n "CN=MyCompany" -len 2048 MyApp.cer -r
pvk2pfx.exe -pvk MyApp.pvk -spc MyApp.cer -pfx MyApp.pfx -po MyPassword
signtool.exe sign /f MyApp.pfx /p MyPassword "Release\WFPSampler.exe"
```
**Use signtool to sign all of WFPSampler solution output (driver, service, ...).**

After sign all output, I've deployed and test on virtual machine (bridge static ip 192.168.1.62), it worked very well. Below image that WFPSampler blocked packet come from Google DNS server (8.8.8.8)
[image]

volehuy1998, 2021-09-29:
> [image]
> 
> WDK: 10.0.22000.1 and 10.0.19041.0 SDK: 10.0.20348.0 and 10.0.19041.0 Visual Studio 2019 Win10: 21h1 build 19043.1237
> 
> Script run result
> I try some things but it not work.
> 
> * Disable [Secure Boot](https://answers.microsoft.com/en-us/protect/forum/all/your-organization-used-windows-defender/ef12b7be-41f5-4b71-a73c-e4f99ba944f7)
> * Turn off all setting in Window Security > App & browser control

Nếu bạn làm theo hướng dẫn trên (https://github.com/microsoft/Windows-driver-samples/issues/661#issuecomment-929869181) mà vẫn có gì trục trặc thì call [facebook](fb.com/volehuy.chingchong) hoặc zalo (0343391748) mình hướng dẫn cho. Mình đã build và chạy thành công.

tqk2811, 2021-09-29:
[image]

Thank You

closed 2021-09-29 by tqk2811

volehuy1998, 2021-09-29:
It work. Nice !!!