microsoft/Windows-driver-samples #662: Scanner: The system cannot find the file specified.

volehuy1998, 2021-10-01:
```
sc create scanner type= kernel binPath= scanner.sys
[SC] CreateService SUCCESS

sc start scanner
[SC] StartService FAILED 2:

The system cannot find the file specified.
```
How to install driver scanner ??

ContegoCode, 2021-11-27:
This is how i fixed it... build the project and in the output there is a avscan.exe and the sys file is in the filter folder
simply copy the avscan.exe into the same folder as .sys driver and right click the .inf file and select install. This will install the driver and you can now run the exe in cmd window.

ContegoCode, 2021-11-27:
unfortunately  I have no idea what this monitors as it shows nothing in the cmd window

closed 2025-11-11 by v-junyli