microsoft/Windows-driver-samples #675: Unable to open include file: "audioengineextensionapo.h"

Lixuya, 2021-11-25:
I installed WDK10.0.19041 and SDK in the installation instructions, but when I run the project it prompts me Unable to open include file: "audioengineextensionapo.h"

Lixuya, 2021-11-25:
Windows-driver-samples\audio\sysvad   sln

JarvisChu, 2022-08-31:
I have the same problem, and I fixed this by using the code version before "audioengineextensionapo.h" was imported, hope to help.

```
git checkout 1ce6f90
git switch -c feat-my-sysvad
```

[image]

[image]

Adonis-Song, 2022-11-30:
Hello. Do you find resolution?

YoungNed, 2022-12-02:
Same here, not sure where to get this header seems not include in the WDK

Adonis-Song, 2022-12-14:
> Hello. Do you find resolution?

Update WDK to the lastest version