microsoft/Windows-driver-samples #677: Installing Sysvad unsuccessful | devcon failed

akashrajkn, 2021-12-01:
I am following the driver (build) and installation process as mentioned here: [SysVAD Virtual Audio Device Driver Sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-driver-samples/sysvad-virtual-audio-device-driver-sample/)

- I am able to build the solution on VS (debug, win32 configuration)
- On running the sample on target computer:
    - I get the following error: `C:\\Tools\\devcon failed`

Any help is appreciated

shilanzhang, 2022-06-30:
I got this error due to   the package in fact is not in current dir.  for you case, are  sure all the files is under c:\sysvadDriver ?

akashrajkn, 2022-06-30:
Thanks for the reply. I will have to check it out and let you know, since it's been a few months

closed 2022-12-03 by jasonknichel