microsoft/Windows-driver-samples #678: [SimpleMediaSourceDriver] Driver signature is invalid

AntumArk, 2021-12-02:
### Problem
I have encountered a problem while trying to add built SimpleMediaSource driver. The driver builds, and all of the four files are present, but I can't seem to be able to add the device.

[image]

### Used commands
`C:\Users\[User]\Windows-driver-samples\general\SimpleMediaSource\x64\Release>pnputil /i /a SimpleMediaSourceDriver.inf
Microsoft PnP Utility

Processing inf :            SimpleMediaSourceDriver.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Total attempted:              1
Number successfully imported: 0

C:\Users\[User]\Windows-driver-samples\general\SimpleMediaSource\x64\Release>cd SimpleMediaSourceDriver

C:\Users\[User\Windows-driver-samples\general\SimpleMediaSource\x64\Release\SimpleMediaSourceDriver>pnputil /i /a SimpleMediaSourceDriver.inf
Microsoft PnP Utility

Processing inf :            SimpleMediaSourceDriver.inf
Adding the driver package failed : A problem was encountered while attempting to add the driver to the store.

Total attempted:              1
Number successfully imported: 0`

Also opening CAT file, gives :
[image]

### Purpose for using this driver
I wish to emulate several cameras, so I could perform test on my application both locally and in CI/CI.

rakendrathapa, 2022-04-28:
Either you need to have MS-approved certificates and sign the driver. That's a big process involved !!

Else, you need to disable signing enforcement on your test machine. 
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/test-signing

AntumArk, 2022-04-28:
Oh, so I guess I won't be able to certificate. Indeed that would be a big process. I am currently working on a different part of a project, but I will try disabling the signing. Thanks!

closed 2022-04-28 by AntumArk