microsoft/Windows-driver-samples #680: Class "Sample" is reserved for use by Microsoft.

lgynju, 2021-12-20:
When I try to build the kmd driver `echo`, it reports 
[image]

my test env is `vs 2019(latest)` + `win 10(lastest)`.
This issue can be produced on my two PC,
Is there any configuration to change?

dechamps, 2021-12-27:
For the record I am hitting the exact same problem:

```
Build started...
1>------ Build started: Project: echo, Configuration: Debug x64 ------
1>Building 'echo' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
1>Stamping x64\Debug\echo.inf
1>Stamping [Version] section with DriverVer=12/27/2021,15.31.30.765
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\AutoSync\echo.inx(16-16): warning 1324: [Version] section should specify PnpLockdown=1.
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\AutoSync\echo.inx(18-18): error 1284: Class "Sample" is reserved for use by Microsoft.
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\AutoSync\echo.inx(30-30): error 1285: Cannot specify [ClassInstall32] section for Microsoft-defined class.
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\AutoSync\echo.inx(64-64): error 1293: Service name 'ECHO' is reserved.
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\AutoSync\echo.inx(90-90): error 1293: Service name 'ECHO' is reserved.
1>Done building project "echo.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

```
1>------ Build started: Project: echo_2, Configuration: Debug x64 ------
1>Building 'echo_2' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
1>Stamping x64\Debug\echo_2.inf
1>Stamping [Version] section with DriverVer=12/27/2021,15.43.42.614
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\DriverSync\echo_2.inx(16-16): warning 1324: [Version] section should specify PnpLockdown=1.
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\DriverSync\echo_2.inx(18-18): error 1284: Class "Sample" is reserved for use by Microsoft.
1>C:\Users\edechamps\Documents\Windows-driver-samples\general\echo\kmdf\driver\DriverSync\echo_2.inx(30-30): error 1285: Cannot specify [ClassInstall32] section for Microsoft-defined class.
1>Done building project "echo_2.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

Visual Studio 2019 16.11.8, WDK 10.0.22000.1.

dechamps, 2021-12-27:
This appears to be a regression in recent WDKs. I just tried it using WDK 10.0.19041.685 and it works.

taochenlove, 2022-01-04:
I have the same problem.what should I do ?

malpka, 2022-01-17:
Same issue exists with Windows 11 EWDK + kmd echo driver

PvPsFinest, 2022-01-21:
> This appears to be a regression in recent WDKs. I just tried it using WDK 10.0.19041.685 and it works.

Anyone know a way to load an older version of the WDK into VS19?
I've tried downloading your version via Chocolatey, but my VS is still saying 10.0.21... 
Also, the Microsoft page I used to install the extension only has the latest WDK versions.

dechamps, 2022-01-21:
I got it from [that page](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#step-2-install-the-wdk), "WDK for Windows 10, version 2004".

HollisTech, 2022-01-21:
Just change the class in the inf file to something other than 'Sample'. (The inf file source  is actually 'echo.inx'.) The only bug here is that the readme ought to be updated.

dechamps, 2022-01-21:
I think a better fix would be to remove "Sample" from the list of Microsoft-reserved classes in the INF stamping tool. What is the point of having a "Sample" class if third-party driver developers can't build *sample* drivers with it?

closed 2025-11-11 by v-junyli