microsoft/Windows-driver-samples #683: maybe stupid question of a beginner

RobertoB74, 2022-01-02:
- i have compiled and installed and 'i don't know' if it work correctly  (i can't use the WPP tracer)
a base driver example .... 
- obviously it's based on a 'usb logic / stack' and in the inf it has a 'USB\....' driver path
- my question is.... like when a driver is installed but no one physical devices are plugged-in the driver icon will be not show in the
  graphical device treeview....

- maybe i need to simulate 'a virtual plug' .... is there some app that simulate a physical usb device?
- or how can i know if my driver is working well without using the windbg or the WPP tracer?

thanks for help

HollisTech, 2022-01-03:
PnP drivers will install with no device present, but will not load until the device specified in the inf file is connected to the system. If you are just trying things out, there are other samples that do not require physical devices. 'echo' for example.

HollisTech, 2022-01-03:
Simulating a usb device is way too complicated to be of any use. You would be better off buying a simple usb dev board instead and use that for experimentation.

RobertoB74, 2022-01-11:
thank you for your kindness ....
but obviously every 'personal DEVICE IO controls' i will couldn't test ....
i will think to a solution with arduino or something other

have a nice day and still thank you