microsoft/Windows-driver-samples #740: Filesys Minifilter Drivers Not Working On Windows 11 (x64)

jeff-simeon, 2022-06-05:
I am building the NameChanger and simrep drivers on Windows 11 x64. They build, install and the services start successfully (having made one edit to add NTamd64 to the DefaultInstall/Uninstall of the inf files. I am building them, targeting x64.

Once started, neither of them appear to do anything. 

For NameChanger, I've created the C:\A\B directory and C:\X directory as described in the instruction.

For simrep, I've created the C:\a\b and C:\x\y directories.

For either example, when I navigate in File Explorer to the user mapping directory, it just opens as normal. No name changing or reparse pointing occurs. I've been testing these each (simrep and NameChanger) independently, just in case one was working and the other was not...but neither seem to do anything.

Please find evidence of the successful instant and service start below:
[image]

[image]

What am I doing wrong here?

Thanks.

ruleil, 2023-04-06:
@jeff-simeon  Have you get this issue fixed? I'm facing the same issue, both NameChanger and SimRep not working on my Windows 10 machine. I've setup kernel debug environments, the registered callbacks(e.g. callback for IRP_MJ_CREATE) are not called at all.

ruleil, 2023-04-07:
I fixed this issue by removing 'return STATUS_FLT_DO_NOT_ATTACH' from NcInstanceSetup which disable mini-filter driver auto-attachment. For SimRep, we need to do similar change to allow instance auto attach. Otherwise, we need to have a user-mode application that can attach mini-filter instance automatically.
[image]

closed 2025-12-17 by v-junyli