microsoft/Windows-driver-samples #746: sysVAD is broken and cannot be loaded

AndreiHarasemiuc, 2022-06-19:
I'm following the APO development docs to the letter and I am successfully building sysVAD on my Windows 10 Vers 21H2 (19044.1645) with Virtual Studio 2022. 

When trying to deploy on my target machine (same machine) I get the following:

[image]

Attempted this on two different machines with the exact same result.
Looking at `setup.api.dev.log`, here's a piece of the output for installing ComponetizedAudioSampleExtension.inf:

`Device not started: Device has problem: 0x27 (CM_PROB_DRIVER_FAILED_LOAD), problem status: 0xc00000263`
The only clue I have about this.

Here's all the `setup.api.dev.log` output for installing the 3 `.inf` files (ComponetizedAudioSample.inf, ComponetizedAudioSampleExtension.inf, ComponetizedApoSample.inf):

```

     cmd: C:\\tools\\devcon  install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample
     ndv: INF path: C:\sysvad\ComponentizedAudioSample.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\MEDIA\0000}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'C:\sysvad\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 10:52:59.688
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     sig:           {_VERIFY_FILE_SIGNATURE} 10:52:59.719
     sig:                Key      = componentizedaudiosample.inf
     sig:                FilePath = c:\sysvad\componentizedaudiosample.inf
     sig:                Catalog  = c:\sysvad\sysvad.cat
!    sig:                Verifying file against specific (valid) catalog failed.
!    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 10:52:59.766
     sig:           {_VERIFY_FILE_SIGNATURE} 10:52:59.766
     sig:                Key      = componentizedaudiosample.inf
     sig:                FilePath = c:\sysvad\componentizedaudiosample.inf
     sig:                Catalog  = c:\sysvad\sysvad.cat
!    sig:                Verifying file against specific Authenticode(tm) catalog failed.
!    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 10:52:59.797
     dvi:           Created Driver Node:
     dvi:                HardwareID   - Root\sysvad_ComponentizedAudioSample
     dvi:                InfName      - c:\sysvad\componentizedaudiosample.inf
     dvi:                DevDesc      - Virtual Audio Device (WDM) - Tablet Sample
     dvi:                Section      - SYSVAD_SA.NT
     dvi:                Rank         - 0x80ff0000
     dvi:                Signer Score - Not digitally signed
     dvi:                DrvDate      - 06/19/2022
     dvi:                Version      - 10.44.0.489
     dvi:           Created Extension Driver Node:
     dvi:                ExtensionID  - {1d641578-54d2-404d-85de-79bb01461ac2}
     dvi:                HardwareID   - Root\sysvad_ComponentizedAudioSample
     dvi:                InfName      - C:\Windows\System32\DriverStore\FileRepository\componentizedaudiosampleextension.inf_amd64_bda3771c83ec23ad\componentizedaudiosampleextension.inf
     dvi:                DevDesc      - Sample Device Extension
     dvi:                Section      - DeviceExtension_Install
     dvi:                Rank         - 0x80ff0000
     dvi:                Signer Score - Not digitally signed
     dvi:                DrvDate      - 06/17/2022
     dvi:                Version      - 14.35.24.441
     dvi:           Created Extension Driver Node:
     dvi:                ExtensionID  - {1d641578-54d2-404d-85de-79bb01461ac2}
     dvi:                HardwareID   - Root\sysvad_ComponentizedAudioSample
     dvi:                InfName      - C:\Windows\System32\DriverStore\FileRepository\componentizedaudiosampleextension.inf_amd64_2382ec4e3184a544\componentizedaudiosampleextension.inf
     dvi:                DevDesc      - Sample Device Extension
     dvi:                Section      - DeviceExtension_Install
     dvi:                Rank         - 0x80ff0000
     dvi:                Signer Score - Not digitally signed
     dvi:                DrvDate      - 06/19/2022
     dvi:                Version      - 7.36.4.367
     dvi:      {Build Driver List - exit(0x00000000)} 10:52:59.938
     dvi:      {DIF_SELECTBESTCOMPATDRV} 10:52:59.954
[... 440 lines cut ...]
     idb:                     {Register Driver Package: C:\Windows\System32\DriverStore\FileRepository\componentizedaposample.inf_amd64_22193fc973923628\ComponentizedApoSample.inf} 10:53:47.637
     idb:                          Created driver package object 'componentizedaposample.inf_amd64_22193fc973923628' in DRIVERS database node.
     idb:                          Created driver INF file object 'oem19.inf' in DRIVERS database node.
     idb:                          Registered driver package 'componentizedaposample.inf_amd64_22193fc973923628' with 'oem19.inf'.
     idb:                     {Register Driver Package: exit(0x00000000)} 10:53:47.637
     idb:                     {Publish Driver Package: C:\Windows\System32\DriverStore\FileRepository\componentizedaposample.inf_amd64_22193fc973923628\ComponentizedApoSample.inf} 10:53:47.637
     idb:                          Activating driver package 'componentizedaposample.inf_amd64_22193fc973923628'.
     cpy:                          Published 'componentizedaposample.inf_amd64_22193fc973923628\componentizedaposample.inf' to 'oem19.inf'.
     idb:                          Indexed 2 device IDs for 'componentizedaposample.inf_amd64_22193fc973923628'.
     sto:                          Flushed driver database node 'DRIVERS'. Time = 0 ms
     sto:                          Flushed driver database node 'SYSTEM'. Time = 16 ms
     idb:                     {Publish Driver Package: exit(0x00000000)} 10:53:47.668
     sto:                     {DRIVERSTORE IMPORT END} 10:53:47.668
     dvi:                          Flushed all driver package files to disk. Time = 16 ms
     sig:                          Installed catalog 'sysvad.cat' as 'oem19.cat'.
     sto:                     {DRIVERSTORE IMPORT END: exit(0x00000000)} 10:53:47.700
     sto:                {Core Driver Package Import: exit(0x00000000)} 10:53:47.700
     sto:           {Stage Driver Package: exit(0x00000000)} 10:53:47.700
     dvs:      {Driver Setup Import Driver Package - exit (0x00000000)} 10:53:47.778
!    dvs:      Unable to find any matching devices.
     dvs: {DrvSetupInstallDriver - exit(00000000)}
<<<  Section end 2022/06/19 10:53:47.793
<<<  [Exit status: SUCCESS]

````

shilanzhang, 2022-06-29:
This is because  the driver is not official signed.
you can deploy it on the  VM by enable test sign .
There are many documents/links talking about how to test  SYSVAD sample.
but this is the only one I can remeber.:
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/test-signing

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers--kernel-mode-

The last one tells you everything.

EmmettTsai, 2022-09-14:
I have the same problem.

My tool and WDK version:
```
Microsoft Visual Studio 2022 Version 17.3.1
Windows Driver Kit 10.0.22621.311
```

Issue can reproduce on:
```
Windows 10 2004 (19041.1110) (Hyper-V virtual machine)
Windows 10 21H2 (19044.1889) (Desktop device)
Windows 11 21H2 (22000.318) (Hyper-V virtual machine)

// I forgot the OS version, these devices are placed in my company, I will update the version at work tomorrow.
Windows 10 ?? (Desktop device)
Windows 11 ?? (Notebook * 2) 
```

Issue **can not** reproduce on:
`Windows 11 22H2 (22621.1) (Hyper-V virtual machine)`

After some test, I found that calling **ExAllocateFromNPagedLookasideList** or **ExFreeToNPagedLookasideList** would cause this issue.
My workaround was to use the old function instead (the "#else" case of (NTDDI_VERSION >= NTDDI_WIN10_NI) in wdm.h).

My workaround code:
https://github.com/microsoft/Windows-driver-samples/commit/527741561f6b7e0e115da26ae728a99cafdc89c8

MikeTMinnick, 2022-10-05:
To solve this, try the following. In Solution Explorer, right-click on the TableAudioSample project and select Properties. Expand Driver Settings and click on General. Select _NT_TARGET_VERSION. In the drop down box, select the Windows version running on your target machine, for example Windows 10.0.22621. Set it for both the Debug and Release configurations. Rebuild the solution and reinstall. 

You may need to update your version of Visual Studio to get support for the latest _NT_TARGET_VERSION.