microsoft/Windows-driver-samples #753: FastFat can sometimes fail FilePositionInformation unnecessarily for read-only media

AshwinTidke, 2022-07-08:
FIC FilePositionInformation should be excluded from the IRP_MJ_SET_INFORMATION check here: https://github.com/microsoft/Windows-driver-samples/blob/d0b7e627fc4aebe6b9c3805a63d0aba4557433ab/filesys/fastfat/verfysup.c#L1707

It should work even for write protected media.

This can cause application failures e.g. opening pdfs using acrobat reader on read-only disks.

Zero3K, 2025-03-01:
It looks like this bug still hasn't been fixed.

msscolee, 2025-11-12:
> @Raj Das ***@***.***>, please share the contact(s) for @microsoft/filesystems<https://github.com/orgs/microsoft/teams/filesystems>.
>
> Thanks,
>
> Scott
>
>
> @msscolee<https://github.com/msscolee> I am sorry for the noise. @microsoft/filesystems<https://github.com/orgs/microsoft/teams/filesystems> was the maintainer team for the FastFat samples, but looks like it's an empty alias now, do you happen to know who the owner is?
>
> -
> You are receiving this email because you were mentioned.

Contact for @microsoft/filesystems<https://github.com/orgs/microsoft/teams/filesystems> should be: ***@***.******@***.***>

Thx,
-Raj

Microsoft365 Running on ReFS Boot Windows11.