microsoft/Windows-driver-samples #761: Minispy only works on one of my computer

Ikaroc-info, 2022-07-21:
Hi everyone! I am trying to develop a new driver based on minispy code. I have successfully install minispy on my windows 11 (Surface computer) But i am trying to install the driver also on a window 10 (HP) and a Windows 11 dev (VM created with hyper-V) but it doesn't work... I have found the base code in this link : https://github.com/microsoft/windows-driver-samples/tree/main/filesys/miniFilter/minispy

after that i build it in Realease mode with visual studio 22 17.1.7 and i tried to install it by right-clicking on the minispy.inf and on install.

After this part I see minispy when i am running the command

`driverquery`

in my computers and VM, but when i am trying to start the service by running in an administrator command prompt

`net start minispy`

I get this result on my VM and my windows 10 computer (translate form french) :

```
error system 127

the procedure is unfindable
```

I don't really know what are the differences and I am feeling kind of lost here... If someone can help me i would be very happy!

fwang2002, 2022-07-28:
I have the same problem.
VS Community 2022, Windows 10.

fwang2002, 2022-08-01:
by using Dependency Walker, find ExAllocateFromNPagedLookasideList and  ExFreeToNPagedLookasideList is not exported.

lyc8503, 2023-03-05:
I also came across the same problem on Windows 10, maybe the cause is same with #806 ?