microsoft/Windows-driver-samples #762: No JobCopiesAllDocuments feature in the xps file generated by XpsDrvSmpl

sylin211, 2022-07-26:
I develop a xps driver based on xpsdrvsmpl. When I choose to print multiple copies of a pdf file, only one copy was printed. I unzip the file and open the print ticket file Job_PT.xml and found that there is no JobCopiesAllDocuments feature (about the number of copies) in it. 

However, I found that the xps file that generated by Microsoft XPS Document Writer contains this feature (See the picture below). 
[image]

Does anyone know why xpsdrvsmpl didn't generate this feature? How do I modify the code to generate it?

elvis-sun, 2022-11-21:
@sylin211 
   I am developing virtual xps driver, I don't know that how to get the xps file content.
   Could you tell me how to get xps file date in the xps drvier?

wstaelens, 2022-11-21:
@sylin211 hopefully somebody will have a look at it as we've seen also issues with XPS but nobody seems to be caring in fixing them:
https://github.com/dotnet/runtime/issues/51929#issuecomment-1166404820 

( @elvis-sun please stop asking in every topic about how to save xps print jobs. Go read the documentation. )