microsoft/Windows-driver-samples #772: "XPSDrv Driver and Filter Sample" Build by VS2019, how to install it?

dybb8999, 2022-08-18:
I try to build “XPSDrv Driver and Filter Sample”, but I meet some error `Error 1297 (NTarm64) Device driver does not install on any devices, use primitive driver if this is intended.` I know it's because of a new primitive driver. So, I fix this error by [Creating a new primitive driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver#legacy-compatibility).
This is inf file after my changed:
```
;
; Copyright (c) 2005  Microsoft Corporation
;
; All rights reserved.
;
; THIS CODE AND INFORMATION IS PROVIDED "AS IS" WITHOUT WARRANTY OF
; ANY KIND, EITHER EXPRESSED OR IMPLIED, INCLUDING BUT NOT LIMITED TO
; THE IMPLIED WARRANTIES OF MERCHANTABILITY AND/OR FITNESS FOR A
; PARTICULAR PURPOSE.
;
; File Name:
;
;    xdsmpl.inf
;
; Abstract:
;
;    XPSDrv sample driver install file
;
[Version]
Signature="$Windows NT$"
Provider=%ProviderString%
ClassGUID={4D36E979-E325-11CE-BFC1-08002BE10318}
Class=Printer
DriverVer=10/17/2008,6.1.6930.0
CatalogFile=XpsDrvSmpl.cat
PnpLockdown=1

;[Manufacturer]
;%ManufacturerName%=Standard,NTx86,NTia64,NTamd64,NTx86.6.0,NTia64.6.0,NTamd64.6.0,NTarm64.6.0

[DefaultInstall.NTx86]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_PRE_VISTA

[DefaultInstall.NTia64]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_PRE_VISTA

[DefaultInstall.NTamd64]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_PRE_VISTA

[DefaultInstall.NTx86.6.0]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_VISTA

[DefaultInstall.NTia64.6.0]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_VISTA

[DefaultInstall.NTamd64.6.0]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_VISTA

[DefaultInstall.NTarm64.6.0]
"XPSDrv Sample Driver" = INSTALL_XDSMPL_FILTERS_VISTA

[DefaultUninstall.NTx86]
LegacyUninstall=1

[DefaultUninstall.NTia64]
LegacyUninstall=1

[DefaultUninstall.NTamd64]
LegacyUninstall=1

[DefaultUninstall.NTx86.6.0]
LegacyUninstall=1

[DefaultUninstall.NTia64.6.0]
LegacyUninstall=1

[DefaultUninstall.NTamd64.6.0]
LegacyUninstall=1

[DefaultUninstall.NTarm64.6.0]
LegacyUninstall=1

[INSTALL_XDSMPL_FILTERS_PRE_VISTA]
CopyFiles=XPSDrvSample,ConfigPlugin,COLORPROFILES
DriverFile=mxdwdrv.dll
PrintProcessor="MS_XPS,filterpipelineprintproc.dll"
ConfigFile=UniDrvUI.dll
HelpFile=UniDrv.HLP
DataFile=XDSmpl.GPD
Include=NTPRINT.INF, MSXPSDRV.INF
Needs=UNIDRV.OEM, XPSGPD.OEM, XPSDRV.OEM

[INSTALL_XDSMPL_FILTERS_VISTA]
CopyFiles=XPSDrvSample,ConfigPlugin,COLORPROFILES
DriverFile=mxdwdrv.dll
ConfigFile=UniDrvUI.dll
HelpFile=UniDrv.HLP
DataFile=XDSmpl.GPD
ICMProfiles=xdwscRGB.icc
CoreDriverSections="{D20EA372-DD35-4950-9ED8-A6335AFE79F0},UNIDRV.OEM", "{D20EA372-DD35-4950-9ED8-A6335AFE79F5},XPSDRV.OEM,XPSGPD.OEM"

[PrinterPackageInstallation.x86]
PackageAware=TRUE
CoreDriverDependencies={D20EA372-DD35-4950-9ED8-A6335AFE79F0}, {D20EA372-DD35-4950-9ED8-A6335AFE79F5}

[PrinterPackageInstallation.amd64]
PackageAware=TRUE
CoreDriverDependencies={D20EA372-DD35-4950-9ED8-A6335AFE79F0}, {D20EA372-DD35-4950-9ED8-A6335AFE79F5}

[PrinterPackageInstallation.ia64]
PackageAware=TRUE
CoreDriverDependencies={D20EA372-DD35-4950-9ED8-A6335AFE79F0}, {D20EA372-DD35-4950-9ED8-A6335AFE79F5}

[PrinterPackageInstallation.arm64]
PackageAware=TRUE
CoreDriverDependencies={D20EA372-DD35-4950-9ED8-A6335AFE79F0}, {D20EA372-DD35-4950-9ED8-A6335AFE79F5}

[DestinationDirs]
DefaultDestDir=66000
COLORPROFILES=66003

[COLORPROFILES]
xdwscRGB.icc
xdCMYKPrinter.icc

[ConfigPlugin]
XDSmpl.ini
XDSmplUI.dll

[XPSDrvSample]
xdsmpl.gpd
xdnames.gpd
xdwmark.gpd
xdbook.gpd
xdcolman.gpd
xdnup.gpd
xdpgscl.gpd
xdwmark.dll
xdcolman.dll
xdbook.dll
xdnup.dll
xdscale.dll
xdsmpl-pipelineconfig.xml

[SourceDisksNames.x86]
1 = %Location%,,
2 = %Location%,,,x86

[SourceDisksNames.ia64]
1 = %Location%,,
2 = %Location%,,,ia64

[SourceDisksNames.amd64]
1 = %Location%,,
2 = %Location%,,,amd64

[SourceDisksNames.arm64]
1 = %Location%,,
2 = %Location%,,,arm64

[SourceDisksFiles]
xdsmpl.gpd                    = 1
xdnames.gpd                   = 1
xdwmark.gpd                   = 1
xdbook.gpd                    = 1
xdcolman.gpd                  = 1
xdnup.gpd                     = 1
xdpgscl.gpd                   = 1
xdsmpl-pipelineconfig.xml     = 1
xdsmpl.ini                    = 1
xdwscRGB.icc                  = 1
xdCMYKPrinter.icc             = 1
xdwmark.dll                   = 1
xdcolman.dll                  = 1
xdbook.dll                    = 1
xdnup.dll                     = 1
xdsmplui.dll                  = 1
xdscale.dll                   = 1

[Strings]
Location="XPSDrv Sample Driver Location"
ManufacturerName="TODO-Set-Manufacturer"
ProviderString = "TODO-Set-Provider"
```

Then build success, I try to install this driver but failed:
[image]

The step as same as  [Build and Install XPS Printer Driver](https://www.youtube.com/watch?v=1XPPsVLr4E8)

I don't know, what different? Just I build this project by VS2019 not VS2017?

I try right click on xdsmpl.inf file and select Install. It's successfully. Why?
[image]

```
Microsoft Visual Studio Professional 2019
版本 16.11.18
VisualStudio.16.Release/16.11.18+32802.440
Microsoft .NET Framework
版本 4.8.09032

已安装的版本: Professional

Visual C++ 2019   00435-30016-55986-AA408
Microsoft Visual C++ 2019

All Force Utf8 (but Java) with BOM   1.2
Force all saved file encoding to Utf8 with BOM, but Java file encoding to Utf8 without BOM

ASP.NET and Web Tools 2019   16.11.110.1451
ASP.NET and Web Tools 2019

Azure 应用服务工具 3.0.0 版   16.11.110.1451
Azure 应用服务工具 3.0.0 版

C# 工具   3.11.0-4.22108.8+d9bef045c4362fbcab27ef35daec4e95c8ff47e1
IDE 中使用的 C# 组件。可能使用其他版本的编译器，具体取决于你的项目类型和设置。

Debugging Tools for Windows   10.0.19030.1000
Integrates the Windows Debugger functionality (http://go.microsoft.com/fwlink/?linkid=223405) in Visual Studio.

IntelliCode 扩展   1.0
IntelliCode Visual Studio 扩展详细信息

Linux 核心转储调试   1.0.9.32413
启用 Linux 核心转储的调试。

Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines

Microsoft MI-Based Debugger   1.0
Provides support for connecting Visual Studio to MI compatible debuggers

Microsoft Visual C++ 向导   1.0
Microsoft Visual C++ 向导

Microsoft Visual Studio VC 软件包   1.0
Microsoft Visual Studio VC 软件包

NuGet 包管理器   5.11.2
Visual Studio 中的 NuGet 包管理器。有关 NuGet 的详细信息，请访问 https://docs.nuget.org/

ProjectServicesPackage Extension   1.0
ProjectServicesPackage Visual Studio Extension Detailed Info

TypeScript Tools   16.0.30526.2002
TypeScript Tools for Microsoft Visual Studio

Visual Basic 工具   3.11.0-4.22108.8+d9bef045c4362fbcab27ef35daec4e95c8ff47e1
IDE 中使用的 Visual Basic 组件。可能使用其他版本的编译器，具体取决于你的项目类型和设置。

Visual F# Tools   16.11.0-beta.21514.6+b6c2c4f53ea3a08fa603ea93d6d2f808a62a21d1
Microsoft Visual F# Tools

Visual Studio Code 调试适配器主机包   1.0
用于在 Visual Studio 中托管 Visual Studio Code 调试适配器的互操作层

Visual Studio Tools for CMake   1.0
Visual Studio Tools for CMake

Windows Driver Kit   10.0.19030.1000
Headers, libraries, and tools needed to develop, debug, and test Windows drivers (msdn.microsoft.com/en-us/windows/hardware/gg487428.aspx)

WiX Toolset Visual Studio Extension   1.0.0.18
WiX Toolset Visual Studio Extension version 1.0.0.18
Copyright (c) .NET Foundation and contributors. All rights reserved.

适用于 Google Test 的测试适配器   1.0
启用带有针对 Google Test 编写的单元测试的 Visual Studio 测试工具。扩展安装目录中提供了使用条款和第三方通知。

适用于 Linux 开发的 Visual C++   1.0.9.32413
适用于 Linux 开发的 Visual C++

通用 Azure 工具   1.10
通过 Azure 移动服务和 Microsoft Azure Tools 提供通用服务。

用于 Boost.Test 的测试适配器   1.0
通过针对 Boost.Test 编写的单元测试启用 Visual Studio 测试工具。扩展安装目录中提供用户条款和第三方通知。
```

dybb8999, 2022-08-22:
我找到了解决方案：
1. 首先将inf文件从项目中移除
2. 正常进行编译
3. 编译完成后，将inf文件复制到输出目录。即dll文件生成路径，package目录。
4. 使用Inf2Cat为package目录创建cat文件
5. 安装驱动时选择该inf即可

closed 2022-08-22 by dybb8999

RoyYL, 2022-11-06:
Hi @dybb8999，
你的 INF 文件是将 [Manufacturer] 去掉了吗？
在使用 Inf2Cat 创建 cat 文件时，传入的参数是什么样子的？
我遇到了如下错误：
Errors:
22.9.1: xdwscrgb.icc in [colorprofiles] of Release\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
……

RoyYL, 2022-11-06:
搞定了，在测试

taianrc, 2022-11-17:
> 搞定了，在测试

Can share how to do it?

Snaping, 2024-04-07:
> 搞定了，在测试

我也遇到了同样的问题，请问怎么解决的？