microsoft/Windows-driver-samples #783: SimpleMediaSource: high CPU usage

pmaciejew, 2022-09-16:
This example takes from 20 to 50% of cpu, where normal  cameras takes 2-5% of cpu. 

Even without frame generator (code below) I observe high cpu usage. 

RETURN_IF_FAILED( sample->GetBufferByIndex( 0, &outputBuffer ) );
RETURN_IF_FAILED( outputBuffer->QueryInterface( IID_PPV_ARGS( &buffer2D ) ) );
RETURN_IF_FAILED( buffer2D->Lock2DSize( MF2DBuffer_LockFlags_Write, &buffer, &pitch, &bufferStart, &bufferLength ) );
RETURN_IF_FAILED( m_FrameGenerator->CreateFrame( buffer, bufferLength, pitch, m_RgbMask ) );
RETURN_IF_FAILED( buffer2D->Unlock2D() );

pmaciejew, 2022-09-18:
Even if I set frame rate to 30 for media type, RequestSample is called too frequently. Should I use a Sleep to slow down frame generation?

Craigsp1022, 2022-10-19:
Check command ratios in all

Hemmikuhschlemur, 2022-10-19:
Task View, Sort CPU Usage, Right Click, Look up in Browser every CPU task with duplicate cpu processes and remove / research and or remove from trash as well if you want to approach from a graphical interface.