microsoft/Windows-driver-samples #801: OEMUNI sample rendering plug-in: DDI hooks and optional COM interface methods not called

PizzaDiavolo, 2022-11-07:
The OEMUNI sample driver does not call user-defined DDI hooks and optional, but implemented IPrintOemUni(2) COM interface methods.

I isolated the OEMUNI sample from the packet, made a Win10Debug target build and installed it on the same test system with Windows 10 Pro 21H2. The build environment is VS2019 16.11.19 with SDK 10.0.22000.0, WDK 10.0.21381.1000, platform toolset used for the DLL is WindowsApplicationForDrivers10.0.

From the debugger I can see that the initialization of the plug-in seems to be fine, the IPrintOemUni2 interface implementation is queried by the Unidriver, its methods like EnableDriver(), EnablePDEV(), GetInfo(), GetImplementedMethod() etc. are called, the DDI hook table is passed to Unidriver.

But these hooks are never called during printing. As far as I understand at least the OEMStartPage() hook should be called. Neither OEMSendPage() nor OEMStartBanding() or OEMNextBand() is called. The only OEM hook function being called is OEMGetGlyphMode() which passes everything back to the Unidriver. This shows that the basic initialization stuff seems to work.

To make things even worse, the implemented IPrintOemUni interface members like FilterGraphics() or ImageProcessing() never get called even if they are announced as implemented in IPrintOemUni::GetImplementedMethod() which is called by the parent.

I know that Unidriver V3 is a rather outdated printer framework but still supported, isn't it?

VictorFang-2024, 2024-07-12:
I want to Know if your question have been solved?
I encountered the same question, but I don't know how to fix it.
Please let me know if you fix it. Thanks.

PizzaDiavolo, 2024-07-15:
Yes, the issue has been solved. Different methods are called at different levels of processing. Some of them are called on initialization, e.g. when the driver is opened, other ones only during job processing, e.g. when printing a document. I was able to trace these calls inside the debugger.

Thanks.

closed 2024-07-15 by PizzaDiavolo