microsoft/Windows-driver-samples #806: Cannot install netvmini driver

RayburnD, 2022-11-23:
Hi, I've successfully compiled the netvmini driver sample, but I cannot get any of the versions to install. Device Manager puts a yelllow icon on the device and throws this error:

_"Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)

{Driver Entry Point Not Found}
The %hs device driver could not locate the entry point %hs in driver %hs."_

In the System Event Viewer, there's an "Application Popup" error with event ID 26 that shows up upon installation, the message is _"netvmini680.sys cannot find ntoskrnl.exe ExFreeToNPagedLookasideList"_

I've tried compiling on Windows 11 as well as on a clean copy of Windows 10 22H2 (separate machines). I've tried using dpinst, pnputil and Device Manager to install it, I've compiled with both VS2019 and 2022 and always get the same error upon installation. I get the error for all 60, 620, 630, and 680 versions.

Is there a trick to getting it installed or is this an issue?

Thanks for your help.

wm1, 2023-01-13:
In latest WDK, several lookaside list API was changed from inline function to exported from ntoskrnl. Thus, if the driver is built targeting latest OS (which is common in samples), it will fail to load in older OS.

```
#if (NTDDI_VERSION >= NTDDI_WIN10_NI)

VOID
ExFreeToNPagedLookasideList (
    _Inout_ PNPAGED_LOOKASIDE_LIST Lookaside,
    _In_ __drv_freesMem(Mem) PVOID Entry
    );

#else  

      // inline functions
```

The fix is to modify the vcxproj to target older OS. In project settings dialog, it is under "driver settings" section:

[image]

rxlop, 2024-04-17:
@RayburnD Did you ever solve this issue?  I am having the exact same issue.  I tried changing target OS version as @wm1 suggested but then netvmin would not compile.  If you were able to solve the issue what was the solution?

RayburnD, 2024-04-17:
@rxlop Yes, I did get it to work. I ended up creating a Windows 10 virtual machine on my Hyper-V box, installed VS2022, Windows 10 SDK, and the Windows Driver Kit on it. After compiling the driver from there I was able to install it on any PC that is in Test Mode. It seemed as if the installations of Windows I was using previously before making my original post weren't clean enough. As soon as I installed Windows 10 on the VM this time, I immediately installed VS2022, WDK and the Windows SDK before updates or anything else.

A few months back I wiped my dev PC and reinstalled the latest Windows 11 ISO along with VS2022, the WDK and SDK on it as well. After that, I was able to compile netvmini and install it on any PC, whereas before wiping Windows, the compiled copy from my PC would throw the errors mentioned in my original post.

My target OS version from the VS Project settings is set to Windows 10 and the platform is Desktop, so in my case I didn't have to select Windows 8.1 to get it to work, unlike suggested above. I'm using version 10.0.22600 of the Windows 10 SDK, this setting is set in Configuration Properties > General. I recommend only installing that specific SDK version and nothing else.

rxlop, 2024-05-04:
@RayBurnD thanks for the suggestions.  I switched to Windows 11 and was able to install the driver.  That allowed me to develop the driver I was working on which was a derivative of netvmini.  However, my new driver still had to work on Windows 10.  After some digging I found that changing the _NT_TARGET_VERSION setting just below the Target OS in VS project settings to a build number earlier than the Windows 10 I was using solves the problem.  My Windows 10 OS build number was 19045.4291.  The drop down selections for the _NT_TARGE_VERSION setting had a bunch of choices I chose 19041 which was the highest number below 19045.  I compiled the driver using that setting and it worked fine on my version of Windows 10.  Thanks again for pointing me in the right direction.