microsoft/Windows-driver-samples #821: [help] fastfat testing

ivanstepanovftw, 2023-01-02:
After deleting `C:/Windows/System32/Drivers/fastfat.sys`, KDNET does not work:
```
PS C:\Program Files (x86)\Windows Kits\10\Debuggers\x64> .\kdnet.exe w11 50030

Failed to enable kernel debugging.  Is secure boot enabled?
Attempting to enable kernel debugging on the {default} OS installation.
Failed to enable debugging on the {default} OS.  Is secure boot enabled?
```

I am testing on Windows VM with Secure Boot disabled. After enabling Secure Boot, KDNET asks to disable it.

ivanstepanovftw, 2023-01-03:
Same in Windows 10

ivanstepanovftw, 2023-01-03:
Going deeper:
```
> bcdedit /debug on
The boot configuration data store could not be opened.
The system cannot find the file specified.
```

ivanstepanovftw, 2025-11-10:
Okay

v-junyli, 2025-11-11:
@microsoft/storage-core can you take a look at this bug? Sorry if I @ the wrong team alias.

msscolee, 2025-11-11:
@Raj Das ***@***.***>, who can look at this from your team?

Thanks,

Scott

You are receiving this email because you are on a team that was mentioned.