microsoft/Windows-driver-samples #876: Fix code scanning alert - Signed overflow check - storage/class/cdrom/src/ioctl.c

lauragonzalezzz, 2023-02-14:
Tracking issue for:
- [ ] https://github.com/microsoft/Windows-driver-samples/security/code-scanning/11

bharsaklemukesh975, 2023-04-14:
Can I work on this?

lauragonzalezzz, 2023-04-14:
Hi @bharsaklemukesh975! Thank you for catching this! This seems to be a stale issue because the cdrom driver sample has been archived from this repository and is no longer published. I'm afraid your effort would likely be wasted in terms of improving this repo (I doubt your PR would even be approved) but I wont deter you from getting your hands on some code to support your learning journey!

bharsaklemukesh975, 2023-04-14:
Hello @lauragonzalezzz , Thanks for getting back to me so quickly and briefly. If you come across any other issues that I can work on instead of this, please give me a shout. Cheers!

closed 2023-04-14 by lauragonzalezzz