microsoft/Windows-driver-samples #921: Add Platform Runtime Mechanism (PRM) sample code

makubacki, 2023-03-09:
[PRM](https://uefi.org/sites/default/files/resources/Platform%20Runtime%20Mechanism%20-%20with%20legal%20notice.pdf) is a feature that allows a subset of platform firmware code to move out of SMM and execute in OS context.

Support for a new PRM OpRegion type was added to the [ACPI 6.4 Specification](https://uefi.org/node/4149).

Functionality in PRM modules can be invoked via the "ACPI-specification defined interface" using the PRM OpRegion or through a "direct interface". While the former is documented in the ACPI Specification, the latter needs sample code.

This issue tracks the addition of that sample code to this repo.

makubacki, 2023-03-09:
@yinghany, I believe you should be the owner?

guliteb, 2023-03-22:
@yinghany Is there any plan to support PRM sample? Thanks.

yinghany, 2023-04-24:
@guliteb The public direct-call interface has been revamped/simplified as part of the next Windows release. Documentation/samples around this interface will be coordinated with the OS release. We will have more information closer to the release date.

guliteb, 2023-08-19:
@yinghany Is there any update on PRM driver sample? Do you have an estimation of the target timeline? Thanks.

closed 2025-12-17 by v-junyli