microsoft/Windows-driver-samples #979: FWPM_LAYER_ALE_CONNECT_REDIRECT_V4: Can't redirect to local proxy

darind, 2023-04-21:
Hi all, I am trying to develop a WFP driver which can be used to redirect outgoing TCP connections to a local proxy server. To better understand the different components involved, I tried running [WFPSampler project](https://github.com/Microsoft/Windows-driver-samples/tree/main/network/trans/WFPSampler) on a new Windows 11 installation:

```
WFPSampler -clean all
WFPSampler.exe -s PROXY -l FWPM_LAYER_ALE_CONNECT_REDIRECT_V4 -iprp 443 -ipp TCP -pra 127.0.0.1 -prp 8866 -plspid 13220 –v
```

After running this command, it seems that the WFPSampler service is successfully invoked through RPC:

```
INFO:  ProxyScenarioExecute : RpcInvokeScenarioProxy() [status: 0]
```

and I can see that the following filter is registered at the ALE_CONNECT_REDIRECT_V4 layer:

[image]

Unfortunately, no traffic is ever reaching the locally running proxy server. If I make an HTTP connection to a remote server, it simply succeeds as if there are no WFP rules.

I am new to kernel development and not sure how can I further debug this and whether the callout function is ever invoked. I never managed to get TraceView working.

Any pointers would be greatly appreciated.

keith-horton, 2024-05-29:
Sorry, this Connect/Redirect does not work between loopback addresses :(

I'm trying to get this fixed.

y-shevchenko, 2025-10-30:
Hi, There is already a fix?