microsoft/Windows-driver-samples #992: SDV-FailDriver-WDM run SDV failed.

bienxanh1901, 2023-06-07:
Hello,
I am testing the sample that demonstrate the SDV tool here: https://github.com/microsoft/Windows-driver-samples/tree/main/tools/sdv/samples/SDV-FailDriver-WDM

But the SDV tool run failed. I checked the `smvbuild.log` log and see the link error as belows:
`error MSB6006: "link.exe" exited with code -1073740791`

The code **-1073740791** is `STATUS_STACK_BUFFER_OVERRUN`.

I am using Visual studio 2022 with WDK version: 10.0.22621755.

Can anyone help me to troubleshoot this issue?

I have attached the log files here:
[smvbuild.log](https://github.com/microsoft/Windows-driver-samples/files/11675575/smvbuild.log)
[smvcl.log](https://github.com/microsoft/Windows-driver-samples/files/11675598/smvcl.log)
[smvlink1.log](https://github.com/microsoft/Windows-driver-samples/files/11675600/smvlink1.log)

Thank you very much!

bienxanh1901, 2023-06-08:
Hi, 
Does anyone have the solution for this issue? 
I have tested with another driver and still cannot generate SDV report.

vvhitesmoke, 2023-09-11:
I've the same :(