microsoft/Windows-driver-samples #1008: fatal  error C1189: #error:  "No Target Architecture"

aszswaz, 2023-07-20:
I get the following error when generating the fastfat driver example:

```
C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\shared\ntdef.h(197,1): fatal  error C1189: #error:  "No Target Architecture"
```

C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\shared\ntdef.h:

```c
#if defined(_AMD64_) || defined(_X86_) || defined(_ARM64EC_)
#define PROBE_ALIGNMENT( _s ) TYPE_ALIGNMENT( ULONG )
#elif defined(_IA64_) || defined(_ARM_) || defined(_ARM64_)

//
// TODO: WOWXX - Unblock ARM. Make all alignment checks ULONG for now.
//

#define PROBE_ALIGNMENT( _s ) TYPE_ALIGNMENT( ULONG )
#elif !defined(RC_INVOKED)
#error "No Target Architecture"
#endif
```

It seems that "RC_INVOKED" is not defined and the compilation fails, how can I solve it?

Windows Version: 11

Windows Kits Version: 10.0.22

sasaki0487, 2023-07-25:
Did you try to include one of the defines (e.g. _X86_, _AMD64_, according to the target you want to build) in your compile option?
Property > c/c++ preprocessor > preprocessor definition

HydraDragonAntivirus, 2024-08-25:
Thanks it's solved problem also if anyone have problem close other tas which you have header error.