microsoft/Windows-driver-samples #1025: Reserved service name errors when building Toaster KMDF samples

forderud, 2023-09-03:
I'm experiencing the following errors when attempting to build the [general\toaster\toastDrv](https://github.com/microsoft/Windows-driver-samples/tree/main/general/toaster/toastDrv) KMDF projects with Visual Studio 2022:
```
>------ Build started: Project: dynambus, Configuration: Debug x64 ------
>Building 'dynambus' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\bus\dynamic\dynambus.inx(64-64): error 1293: Service name 'dynambus' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\bus\dynamic\dynambus.inx(75-75): error 1293: Service name 'dynambus' is reserved.

>------ Build started: Project: StatBus, Configuration: Debug x64 ------
>Building 'StatBus' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\bus\static\statbus.inx(64-64): error 1293: Service name 'statbus' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\bus\static\statbus.inx(75-75): error 1293: Service name 'statbus' is reserved.

>------ Build started: Project: filter (Kmdf\Filter\Generic\filter), Configuration: Debug x64 ------
>Building 'filter' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\filter\filter.inx(76-76): error 1293: Service name 'ToasterFilter' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\filter\filter.inx(77-77): error 1293: Service name 'wdffeatured' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\filter\filter.inx(96-96): error 1293: Service name 'ToasterFilter' is reserved.

>------ Build started: Project: filter (Kmdf\Filter\Sideband\filter), Configuration: Debug x64 ------
>Building 'filter' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\filter\filter.inx(76-76): error 1293: Service name 'ToasterFilter' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\filter\filter.inx(77-77): error 1293: Service name 'wdffeatured' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\filter\filter.inx(96-96): error 1293: Service name 'ToasterFilter' is reserved.

>------ Build started: Project: wdffeatured, Configuration: Debug x64 ------
>Building 'wdffeatured' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\func\featured\wdffeatured.inx(72-72): error 1293: Service name 'wdffeatured' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\func\featured\wdffeatured.inx(82-82): error 1293: Service name 'wdffeatured' is reserved.

>------ Build started: Project: wdfsimple, Configuration: Debug x64 ------
>Building 'wdfsimple' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\func\simple\wdfsimple.inx(75-75): error 1293: Service name 'wdfsimple' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\func\simple\wdfsimple.inx(85-85): error 1293: Service name 'wdfsimple' is reserved.

>------ Build started: Project: wdftoastmon, Configuration: Debug x64 ------
>Building 'wdftoastmon' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\toastmon\wdftoastmon.inx(22-22): error 1284: Class "Sample" is reserved for use by Microsoft.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\toastmon\wdftoastmon.inx(40-40): error 1285: Cannot specify [ClassInstall32] section for Microsoft-defined class.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\toastmon\wdftoastmon.inx(62-62): error 1293: Service name 'wdfToastMon' is reserved.
>C:\Windows-driver-samples\general\toaster\toastDrv\kmdf\toastmon\wdftoastmon.inx(72-72): error 1293: Service name 'wdfToastMon' is reserved.
```

It would be great if your could fix these errors so that the samples build out-of-the-box. Alternatively, clearly document how to rename the services. The reason for asking is that it's unclear to me if it's sufficient to change the service name, or if the SYS and INX/INF files also need to be renamed.

naveensadashiviah, 2023-09-27:
Even after renaming all the ones for which the errors were reported, there's this error:

`Severity	Code	Description	Project	File	Line	Suppression State
Error	1284	Class "Sample" is reserved for use by Microsoft.	wdftoastmon	C:\Users\nBase2\Git\repos\Windows-driver-samples\general\toaster\toastDrv\kmdf\toastmon\wdftoastmon.inx	22	
`

And if I change "Sample" to "Sample1", it reports this error:

`Severity	Code	Description	Project	File	Line	Suppression State
Error	1280	Class name and ClassGuid mismatch, expecting Class "Sample" for ClassGuid "{78A1C341-4539-11d3-B88D-00C04FAD5171}".	wdftoastmon	C:\Users\nBase2\Git\repos\Windows-driver-samples\general\toaster\toastDrv\kmdf\toastmon\wdftoastmon.inx	22	
`

Basically, we need more details to overcome this error.

forderud, 2023-09-27:
> And if I change "Sample" to "Sample1", it reports this error:
>
> `Severity Code Description Project File Line Suppression State Error 1280 Class name and ClassGuid mismatch, expecting Class "Sample" [...]

`Sample` is a well-known class associated with `ClassGuid={78A1C341-4539-11d3-B88D-00C04FAD5171}`. You'll therefore probably need to also change `ClassGuid` if changing `Class`. This is documented on https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings

v-junyli, 2025-11-11:
@microsoft/coreos-buses-and-wdf