microsoft/Windows-driver-samples #1041: minifilter inf files do not comply with infverify or the currrent primitive driver inf requirements.

HollisTech, 2023-10-18:
All the sample minifilter driver inf files fail infverify with the latest VS2022 WDK.
For example, the ctx minifilter has the following errors:
Validating ctx.inf
  ERROR(1420) in ...\Ctx\x64\Release\ctx.inf, line 0: [DefaultInstall]-based INF cannot be processed as Primitive.
  WARNING(1324) in ...\Ctx\x64\Release\ctx.inf, line 8: [Version] section should specify PnpLockdown=1 to prevent external apps from modifying installed driver files.
  ERROR(1421) in ...\Ctx\x64\Release\ctx.inf, line 25: Section [DefaultInstall] should have an architecture decoration.
  ERROR(1421) in ...\Ctx\x64\Release\ctx.inf, line 36: Section [DefaultUninstall] should have an architecture decoration.
  WARNING(1205) in ...\Ctx\x64\Release\ctx.inf, line 72: Section [MiniFilter.DriverFiles] referenced from DelFiles and CopyFiles directive.
  INF is NOT VALID

HollisTech, 2024-03-14:
Still broken. It's only been open since october '23.

NeoAdonis, 2024-06-04:
@zlockard Can you help me to tell if the issues with filesys/miniFilter/ctx/ctx.inf still remain, or were they successfully fixed in the latest release?

zlockard, 2024-08-20:
This has been resolve

closed 2024-08-20 by zlockard