microsoft/Windows-driver-samples #1056: ComponentizedApoSample.inx uses incorrect AudioEngine location

namelessmasses, 2023-11-22:
In the sample .inx, 

https://github.com/microsoft/Windows-driver-samples/blob/6971eb69270c614964808f0409fe0b479da51270/audio/sysvad/TabletAudioSample/ComponentizedApoSample.inx#L59

shouldn't the AudioEngine location be 

`HKR\Classes\AudioEngine\...`

kkotte, 2023-11-30:
This is by design. Under HKR, `CLSID` and `AudioEngine` are not peers.

thxkiwi, 2023-11-30:
Thanks for the response. 

Just curious about that design. Do you have any knowledge about the change?

I'm specifically wondering 

1. Why change them from being peers?
1. What is it that actually has knowledge of how to find the CLSID and AudioEngine properties under the software key? Will this only work when using AudioProcessingObject setup class? I.e., if these are in the base driver INF (Media setup class) then the audio engine won’t find them.

kkotte, 2023-11-30:
HKR entries are saved in a location relative to the device, and COM registrations relative to a device have a different layout for registrations.

Its the COM subsystem that has knowledge of how to find the COM registration. The Audio stack tells COM which device is of interest when searching for these registrations. Currently, the audio stack will instruct COM to look for COM registrations relative to only the AudioProcessingObject setup class (and not the MEDIA class).

namelessmasses, 2023-11-30:
Closed as I have the primary question(s) answered. If anyone has information how long existing HKCR INFs will be accepted by pnputil/PnPManager please comment.

closed 2023-11-30 by namelessmasses